autowarefoundation/AWSIM-Labs #11: Let user manipulate simulation traffic through UI

mozhoku, 2024-03-11:
Hi!
I'm planning to add traffic control from the ESC menu. With this, users will be able to reset, stop, restart the traffic in the simulation. They'll also be able to change the seed etc. We can also a shortcut key to start/stop the traffic. 

Please give me your thoughts and suggestions about this addition. 🤗

doganulus, 2024-03-11:
Is such traffic control available programmatically? Via ROS topics or using some other way?

mozhoku, 2024-03-11:
> Is such traffic control available programmatically? Via ROS topics or using some other way?

Not that I'm aware of. But it should be possible.

xmfcx, 2024-03-11:
Let's skip "Shortcut Key" option for this issue, we can consider hotkeys in an issue dedicated to hotkey management in general.

mozhoku, 2024-03-28:
- https://github.com/autowarefoundation/AWSIM/pull/17

Changes are merged. Closing this issue.

closed 2024-03-28 by mozhoku