autowarefoundation/AWSIM-Labs #15: Release the next version of AWSIM binary with URP

xmfcx, 2024-03-13:
## Blocked by

- https://github.com/autowarefoundation/AWSIM/issues/12

## Definition of done

- [ ] Produce binaries for both scenario simulation and default AWSIM
- [ ] Test them
- [ ] Upload them
- [ ] Update the documentation with these binaries

lchojnack, 2024-04-16:
I've started to work on this task. During set up I've encountered the issue described [here](https://github.com/autowarefoundation/AWSIM/issues/49). I think that the issue blocks this task.

xmfcx, 2024-04-17:
@lchojnack we have also merged the latest updates to the documentation with:
- https://github.com/autowarefoundation/AWSIM/pull/42

**Updated documentation page:** https://autowarefoundation.github.io/AWSIM/main/

We are now waiting for the release, feel free to ask if you have any questions for this task.

lchojnack, 2024-04-18:
@xmfcx 
I've produced the AWSIM binaries and here are my observations

### AWSIM binary test
I've tested the binary with [Autoware on tag 2024.04](https://github.com/autowarefoundation/autoware/tree/2024.04). 

The binary works well, but I've noticed that green traffic lights are not detected well. 
I've compared the result with the HDRP version of AWSIM to check if it was caused by the rendering pipeline. I think that changing to URP reduces Autoware traffic light detection performance. 

**Question:** Is the not good green light detection acceptable for this release?

| URP (AWSIM Labs) | HDRP  ([AWSIM v.1.2.1](https://github.com/tier4/AWSIM/releases/tag/v1.2.1)) |
| --- | --- |
| <video src="https://github.com/autowarefoundation/AWSIM/assets/8416175/3d5f4225-c599-4f04-a2b0-db7562801eed" width="500" /> | <video src="https://github.com/autowarefoundation/AWSIM/assets/8416175/182430a6-1789-4f43-ab99-c3e90694a983" width="500" /> |

### AWSIM ss2 binary test

I've notice that the AWSIM labs are not applicable for ss2 scene.

**Question:** Is it acceptable for this release or just wait for fix?

| AWSIM bin | AWSIM ss2 bin |
| --- | --- |
| [image] | [image] |

### Github release
**Questions:**
- Could you give me some more permission for [creating/manages releases for the repo](https://docs.github.com/en/repositories/releasing-projects-on-github/managing-releases-in-a-repository#about-release-management)? 

- To be sure, What version is this realese? 1.2.0?

xmfcx, 2024-04-19:
Thanks @lchojnack 

## Traffic light problem

Before release, let's wait for this issue to be resolved:

- https://github.com/autowarefoundation/AWSIM/issues/55

## Scenario simulator

For this release, let's keep scenario simulator out, we can do it in the next release.

## Permissions

I've added you to the repository with Write permissions, please let me know if it doesn't work for you.

## Version

This will be the `1.0.0` for the Autoware Labs.

lchojnack, 2024-04-19:
@xmfcx 

If the release will be `1.0.0`, the version in left-bottom corner of UI shouldn't also point to `1.0.0` instead `1.2.0`?

[image]

xmfcx, 2024-04-24:
I will release the current state with 3 lidars as a beta version `1.0.0-beta` along with
- https://github.com/autowarefoundation/autoware/issues/4659

To update relevant readme files and instructions.

When
- https://github.com/autowarefoundation/AWSIM/pull/57

is merged, let's to the 1.0.0 release.

lchojnack, 2024-05-09:
@mozhoku @xmfcx 
Ok, I see what could be wrong. The Graphy is disabled on the scene by default.
Please let me know, Is it expected or is it a bug?

[image]

mozhoku, 2024-05-09:
@lchojnack 
It should be enabled by default. I've forgot to turn it on at some point, my bad. Could you please turn it on? :pray:

lchojnack, 2024-05-09:
@mozhoku 
Thank you for the clarification. Yes, I will enable it.

lchojnack, 2024-05-09:
@xmfcx 
Release v1.0.0 is ready https://github.com/autowarefoundation/AWSIM/releases/tag/v1.0.0

xmfcx, 2024-05-10:
@brkay54 could you test this v1.0.0 binary against the latest main branch of the Autoware?

xmfcx, 2024-05-10:
Test video with AWSIM Labs 1.0.0: https://www.youtube.com/watch?v=GPow8YPgi1s

Thanks!

closed 2024-05-10 by xmfcx