autowarefoundation/AWSIM-Labs #51: Turn on 2 side lidars too

xmfcx, 2024-04-16:
In the AWSIM Autoware configuration and in the simulation, there are 2 side lidars. With the lidar on top, there are 3 VLP16s on the vehicle.

In the old HDRP, due to performance issues side lidars were turned off.

Let's turn them back on.

Also @brkay54 tested the side lidars and will share some problems they have.

Probably if we just move(translate) them slightly to the sides, the problems will be fixed.

brkay54, 2024-04-18:
**The current left and right lidar output is shown here. My concerns are:**
- side lidars are not symmetrical
- they can not see the front of the vehicle

[image]
[image]
[image]

- We made some draft changes, the points on the vehicle were increased but the side lidars can see the front of vehicle. It is just a draft setup we can improve it.

[image]
[image]
[image]

mozhoku, 2024-04-18:
Okay I've made some changes and opened a PR on the autoware_individual_params repo here:
- https://github.com/autowarefoundation/autoware_individual_params/pull/47

In the AWSIM side I will update the lidars and will look like this:

Before:
[image]

After:
[image]

closed 2024-04-24 by xmfcx