autowarefoundation/AWSIM-Labs #52: Dynamic sensor placement

xmfcx, 2024-04-16:


xmfcx, 2024-04-17:
@SujayAmberkar will work on this issue.

SujayAmberkar, 2024-04-19:
Hi @xmfcx,
**Question**: Place the sensors dynamically means, setting the transforms of these highlighted game objects(in this image) automatically after I press start(in Game mode) or in editor mode? Or am I missing something?
[image]

**Question**: Were the sensor transforms manually inputted and placed under the sensor_kit_base_link game object?

**Doubt**: The values that will be used to determine the transforms will be in a URDF file like this one https://github.com/RobotecAI/awsim_sensor_kit_launch/tree/main/awsim_sensor_kit_description/urdf

Thanks

xmfcx, 2024-04-23:
> **Question:** Place the sensors dynamically means, setting the transforms of these highlighted game objects(in this image) automatically after I press start(in Game mode) or in editor mode? Or am I missing something?

Yes. Additionally we can make a unity tool to import before build time too.

> **Question:** Were the sensor transforms manually inputted and placed under the sensor_kit_base_link game object?

Previously yes. But with this change, we should be able to point to a folder.

The transforms can be read from here, it makes more sense as they are used by the Autoware too: https://github.com/autowarefoundation/autoware_individual_params/tree/main/individual_params/config/default/awsim_sensor_kit

SujayAmberkar, 2024-04-23:
Ok, so I should place this folder named "awsim_sensor_kit" somewhere(where?) in the Unity project -> search this folder for the yaml file -> convert ROS transforms to Unity transforms -> Assign the transforms to the game objects, correct?

 This will be an editor tool, since initially if a developer is following this document to Add sensors https://autowarefoundation.github.io/AWSIM/main/Components/Vehicle/AddNewVehicle/AddSensors/ they'll create just the game objects manually and use the editor tool to assign transforms automatically.

xmfcx, 2024-04-24:
> Ok, so I should place this folder named "awsim_sensor_kit" somewhere(where?) in the Unity project -> search this folder for the yaml file -> convert ROS transforms to Unity transforms -> Assign the transforms to the game objects, correct?

The user can specify the location with a file window or they can enter the file path.

SujayAmberkar, 2024-04-30:
[image]

Progress so far/Approach taken: 
- An editor tool to read yaml file from a given path. (Done)
- Convert ROS transform to unity. (to-do)
- Find the game object mentioned in the YAML file. (bug: only applying transforms to the parent object)
- Apply transforms. (done)

esteve, 2024-04-30:
You can convert xacro files to URDF files with the following command:

```sh
ros2 run xacro xacro -o filename.urdf filename.xacro
```

xmfcx, 2024-04-30:
https://github.com/autowarefoundation/awsim_labs_sensor_kit_launch/tree/main/awsim_labs_sensor_kit_description/urdf

@xmfcx to look into this.

SujayAmberkar, 2024-05-05:
@xmfcx do we have some sample urdf file. I tried this command` ros2 run xacro xacro -o filename.urdf filename.xacro ` but it produced an empty urdf file with some default text.

mitsudome-r, 2024-05-07:
@SujayAmberkar This should be the file you should work with: https://github.com/autowarefoundation/autoware.universe/blob/main/launch/tier4_vehicle_launch/urdf/vehicle.xacro

You should specify `awsim_labs_sensor_kit` for sensor model and `sample_vehicle` for vehicle_model as arguments below:
``` 
  <xacro:arg name="vehicle_model" default=""/>
  <xacro:arg name="sensor_model" default=""/>
```

You can modify them locally as a starting point.

mitsudome-r, 2024-05-08:
1. Build Autoware by following You would have to build Autoware by following the instruction in https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/.

2. Modify L3-4 in `$HOME/autoware/src/universe/autoware.universe/launch/tier4_vehicle_launch/urdf/vehicle.xacro` as the following:
```
  <xacro:arg name="vehicle_model" default="sample_vehicle"/>
  <xacro:arg name="sensor_model" default="awsim_labs_sensor_kit"/>
```

3. Run the following commands:
```
source $HOME/autoware/install/setup.bash
ros2 run xacro xacro -o awsim_vehicle.urdf $HOME/autoware/src/universe/autoware.universe/launch/tier4_vehicle_launch/urdf/vehicle.xacro
```

I'm also attaching the generated file just in case: [awsim_vehicle.zip](https://github.com/autowarefoundation/AWSIM/files/15248870/awsim_vehicle.zip)

SujayAmberkar, 2024-05-13:
Hi, I've made a pull request https://github.com/autowarefoundation/AWSIM/pull/72

The method I follow is:
Read the URDF -> Find game objects in the unity scene by names of <link> -> Extract transforms from <origin> under <link> or <visual> or <inertial> tags -> Convert the rotations and positions -> Apply them to the game objects

There are a few issues with this:
- The first sensor named "VelodyneVLP16" under "sensor_kit_base_link" is not mentioned in the URDF file. Should it be velodyne_top?
- Transforms in URDF are different than the currently assigned transforms, so when the sensors are placed automatically, they are not on the car. For example, the position of base_link is [0,0.04,0] but in the URDF it's [1.395, 0, 0] (ros2 transforms)

[image]

closed 2024-05-21 by SujayAmberkar