autowarefoundation/AWSIM-Labs #65: Add feature to spawn NPC from RVIZ

mitsudome-r, 2024-04-30:


SujayAmberkar, 2024-05-28:
I'm getting this message when I listen to the /simulation/dummy_perception_publisher/object_info,
here the position is around 81000 & 49000 for x and y whenever I create a dummy NPC in rviz. 
1. Should I be using this position?
2. Should it be converted?

```
header:
  stamp:
    sec: 1843
    nanosec: 957748583
  frame_id: map
id:
  uuid:
  - 146
  - 88
  - 184
  - 128
  - 4
  - 100
  - 136
  - 211
  - 81
  - 227
  - 181
  - 129
  - 139
  - 131
  - 198
  - 67
initial_state:
  pose_covariance:
    pose:
      position:
        x: 81383.5859375
        y: 49919.77734375
        z: 0.0
      orientation:
        x: 0.0
        y: 0.0
        z: 0.2974268081526713
        w: 0.95474462229023
    covariance:
    - 0.0008999999845400453
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0008999999845400453
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0008999999845400453
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.007615434937179089
  twist_covariance:
    twist:
      linear:
        x: 3.0
        y: 0.0
        z: 0.0
      angular:
        x: 0.0
        y: 0.0
        z: 0.0
    covariance:
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
  accel_covariance:
    accel:
      linear:
        x: 0.0
        y: 0.0
        z: 0.0
      angular:
        x: 0.0
        y: 0.0
        z: 0.0
    covariance:
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
    - 0.0
classification:
  label: 1
  probability: 1.0
shape:
  type: 0
  footprint:
    points: []
  dimensions:
    x: 4.0
    y: 1.7999999523162842
    z: 2.0
max_velocity: 33.29999923706055
min_velocity: -33.29999923706055
action: 0

```

mozhoku, 2024-05-28:
@SujayAmberkar you can check out this. 

https://github.com/autowarefoundation/AWSIM-Labs/blob/a99f67e1261df08d405de16883d8e000a1140485/Assets/AWSIM/Scripts/Sensors/Gnss/GnssSensor.cs#L73

I'm not sure if it does anything else with the coordinate values.

xmfcx, 2024-06-11:
Message should be added as a custom message:
- https://github.com/autowarefoundation/autoware.universe/tree/36b0b2bde9f49c2d1700266fcb50afff12d618b9/simulator/dummy_perception_publisher/msg

SujayAmberkar, 2024-06-16:
@mozhoku I want to create a new custom message but I'm unsure if it'll cause any problems. And you're working on an issue of replacing older messages with new ones so I wanted to know what I should take care of while creating a new custom message. By reading that issue I could only understand that it's something related to the change of names.

mozhoku, 2024-06-24:
@SujayAmberkar if you are adding a new custom message it shouldn't affect anything. I will be replacing the current message types used by the simulation so that it can communicate with the main autoware_universe branch after the msg rp. 
For example: `autoware_auto_perception_msgs` => `autoware_perception_msg`.

SujayAmberkar, 2024-07-08:
@xmfcx Hello, extremely sorry for the delay. I'll try to finish this as soon as possible. I'm facing a problem here. After adding a custom message in Unity and writing the following code I'm not able to use the message. Here I'm trying to print the message in the callback function but nothing's getting printed. I'm not getting any errors saying "invalid message type" so the message type has been successfully installed. When I run AWSIM and check the terminal to see if there are any subscribers, it shows 4 publishers and 1 subscriber. 
```cs
public class RVIZNPCSpawner : MonoBehaviour
    {
        [SerializeField] string NPCSpawnerTopic = "/simulation/dummy_perception_publisher/object_info";

        // Subscriber
        ISubscription<dummy_perception_publisher.msg.Object> NPCSpawnerSubscriber;

        void Start()
        {
            // Define default QoS settings
            QoSSettings qosSettings = new QoSSettings
            {
                ReliabilityPolicy = ReliabilityPolicy.QOS_POLICY_RELIABILITY_RELIABLE,
                DurabilityPolicy = DurabilityPolicy.QOS_POLICY_DURABILITY_TRANSIENT_LOCAL,
                HistoryPolicy = HistoryPolicy.QOS_POLICY_HISTORY_KEEP_LAST,
                Depth = 1
            };
            
            // Initialize the ROS2 node and create the subscription
            NPCSpawnerSubscriber = SimulatorROS2Node.CreateSubscription<dummy_perception_publisher.msg.Object>(
                NPCSpawnerTopic, NPCSpawnerCallback, qosSettings.GetQoSProfile());
        }

        /// <summary>
        /// Callback method to handle incoming messages
        /// </summary>
        /// <param name="msg">Received Object message</param>
        void NPCSpawnerCallback(dummy_perception_publisher.msg.Object msg)
        {
            Debug.Log(msg);
        }

        void OnDestroy()
        {
            // Clean up the subscription when the GameObject is destroyed
            SimulatorROS2Node.RemoveSubscription<dummy_perception_publisher.msg.Object>(NPCSpawnerSubscriber);
        }
    }
```
[image]

SujayAmberkar, 2024-07-09:
Solved the problem in the previous comment. Here's a video demo of the work so far,
NPC Vehicles and Pedestrians are spawned properly but vehicles do not move.
[RvizSpawning.webm](https://github.com/autowarefoundation/AWSIM-Labs/assets/38327075/df624bfe-e15c-454f-b3a3-33503cb17268)

SujayAmberkar, 2024-07-12:
@xmfcx I've added a feature to make the vehicle move in its forward direction once it's spawned. But here's the problem, since we want to remove the rigid body component the vehicle goes into the ground after moving some distance maybe because the ground is not levelled. Is there a reason for removing the rigid body?
Initial pose:
[image]

After some time:
[image]

xmfcx, 2024-07-12:
Why remove the rigid body? I've suggested removing the colliders. Or colliders could have their layers set accordingly https://docs.unity3d.com/Manual/LayerBasedCollision.html

SujayAmberkar, 2024-07-12:
Ahh ok, my bad. I got confused.

SujayAmberkar, 2024-07-16:
Features Added:

- Spawn NPC pedestrians and vehicles. Vehicles move at constant velocity only in their forward direction.
- NPC Vehicles only collide with the ground.
- Specify the velocity of the vehicles in Autoware(in Tool Properties).
- Random vehicles will be spawned for smaller cars and a truck will be spawned as a large vehicle.(more vehicles can be added)
- Despawn after some time automatically.
- Despawn all NPCs at one from Autoware.

https://github.com/user-attachments/assets/79b67a5c-9c50-4333-8237-658994f09ae8

SujayAmberkar, 2024-07-16:
Here is the draft PR https://github.com/autowarefoundation/AWSIM-Labs/pull/118

closed 2024-07-19 by SujayAmberkar