autowarefoundation/AWSIM-Labs #73: Publish traffic lights signals for headless

mozhoku, 2024-05-15:
There should be a publisher ready for it if I'm not mistaken. Need to check to see if it works properly. If not, will fix it or make a new one.

lchojnack, 2024-05-16:
@mozhoku 

I've implemented the V2I feature for the AWSIM. More details in the documentation [here](https://autowarefoundation.github.io/AWSIM/main/Components/Environment/V2I/)

If you want, I can take care of this task and enable the V2I feature.

mozhoku, 2024-05-16:
@lchojnack 
Thank you, that would be great! I'll assign this task to you.

lchojnack, 2024-05-20:
@mozhoku @xmfcx 

The V2I feature can be easily added by enabling it in the scene.

[image]

I'm wondering if we want to enable it by default or if we want a feature that allows the users to enable/disable V2I by UI. What do you think?

mitsudome-r, 2024-05-21:
We can enable this feature as default for now, and later create toggle buttons as a separate PR.

closed 2024-05-27 by mozhoku