autowarefoundation/AWSIM-Labs #85: AWSIM Crashed

DarkSide-reza, 2024-05-30:
hi guys 
i downloaded the AWSIM Lab And I opened it in Unity Hub and its run ok
but when I tried to start testing and building my own project in AWSIM from its training, unity crashed
and its happened a lot and without any data about the reason ,for example when i try to add component suddenly unity crashed and i have to open project again and very soon crashed again , there is any way to solve it ?

mozhoku, 2024-05-30:
@DarkSide-reza few questions:

1) Does it crash when opening new windows like build menu etc.?
2) When adding component do you add with "mouse click" or "press enter"?

**In case 1;**
Try to disable AutoGenerateLighting option. Sometimes it defaults to CPU for baking. This was causing issues on my potato PC.
You can turn it off from the Light icon in the bottom corner of the editor window. Clicking it opens Light Settings window. In the bottom of the window you can toggle "Auto Generate" option.
[image]

Or you can open your unity scene file with a text editor and change "m_GIWorkflowMode" from 0 to 1.

**In case 2;**
I don't know the reason. In Ubuntu it crashes the editor sometimes when you add the component by pressing "enter". I've seen this issue on multiple PC's.

If it's caused by auto light baking, we can disable it by default later on.

DarkSide-reza, 2024-06-01:
> Hello @DarkSide-reza
> 
> Few questions:
> 
>     1. Does it crash when opening new windows like build menu etc.?
> 
>     2. When adding component do you add with "mouse click" or "press enter"?
> 
> 
> **In case 1;** Try to disable AutoGenerateLighting option. Sometimes it defaults to CPU for baking. This was causing issues on my potato PC. You can turn it off from the Light icon in the bottom corner of the editor window. Clicking it opens Light Settings window. In the bottom of the window you can toggle "Auto Generate" option. You need to do it as soon as the editor loads though. Otherwise it's gonna crash again when opening the Lighting window. ![image](https://private-user-images.githubusercontent.com/33583972/335197143-d6533356-902d-4201-968c-19a3d860a6d9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyMjQwMTEsIm5iZiI6MTcxNzIyMzcxMSwicGF0aCI6Ii8zMzU4Mzk3Mi8zMzUxOTcxNDMtZDY1MzMzNTYtOTAyZC00MjAxLTk2OGMtMTlhM2Q4NjBhNmQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjAxVDA2MzUxMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmMDgwNzc1NzE1NDU4YWI2NGM0MjZlNTk4YzIxMDVmYmJlODJlZGQwMWVhNWYyZTdhZmI4M2Y5MzBlYWIzYzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.7Zzf8koVcV9tN_LyIv8KZ4LoKBstV7G15PnSmpJmV1g)
> 
> Or you can simply open your unity scene file with a text editor and change "m_GIWorkflowMode" from 0 to 1.
> 
> **In case 2;** I don't know the reason. In Ubuntu it crashes the editor sometimes when you add the component by pressing "enter". I've seen this issue on multiple PC's.
> 
> If it's caused by auto light baking, we can disable it by default later on.

DarkSide-reza, 2024-06-01:
thanks for ur help i turned off AutoGenerateLighting but it didn't work
its crashed when im clicking, like add component or when i want to export a package 
and its happened only in awsim project not in other unity projects
i read about unity (Editor.log) before 
how can i use it for solving the problem ?

mozhoku, 2024-06-03:
Could you give me more detailed information about your system?
For example OS, Nvidia driver version etc.

DarkSide-reza, 2024-06-03:
sure 
linux ubuntu 22.04 / 
core i7 12Gen /
RTX 3060 6g / driver 545 /
16G ram /
also im unsing unity 2022.3.21f1 the same version of awsim lab 

and its the editor log for crashing :
=================================================================
	Native Crash Reporting
=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.

	Telemetry Dumper:
=================================================================
Thread 0x763354ffc640 may have been prematurely finalized* Assertion at mono-threads.c:702, condition `info' not met, function:mono_thread_info_current, 
An error has occured in the native fault reporting. Some diagnostic information will be unavailable.

closed 2024-07-06 by DarkSide-reza