autowarefoundation/AWSIM-Labs #88: Migrate to autoware_msgs and release

xmfcx, 2024-06-04:
Mega message migration discussion post:
- [Blocking the Main Branch of Autoware Universe due to the Migration of autoware_msgs #4803](https://github.com/orgs/autowarefoundation/discussions/4803#discussioncomment-9645234)

Migration is done, now we should update the AWSIM Labs to use the new messages.

We should be updating this folder:
- https://github.com/autowarefoundation/AWSIM-Labs/tree/main/Assets/Ros2ForUnity/Plugins

Using these instructions:
- https://autowarefoundation.github.io/AWSIM-Labs/main/Components/ROS2/AddACustomROS2Message/

Tasks:
- [ ] Remove all references to `autoware_auto_x_msgs`
- [ ] Add new `autoware_x_msgs` as custom messages
- [ ] Update the code references (pubs and subs) to use the new messages
- [ ] Test against the Autoware with the demo tutorial
- [ ] Update the documentation if necessary
- [ ] Generate a build and release with all the changes up to this PR.

lchojnack, 2024-06-05:
@mozhoku 
FIY I'm working on a similar migration for [AWSIM](https://github.com/tier4/AWSIM/) on branch [feature/autoware_msgs](https://github.com/tier4/AWSIM/commits/feature/autoware_msgs/) (there is no PR so far)

If you want you can cherry-pick what you think is useful.

mozhoku, 2024-06-05:
@lchojnack Thanks!

Also I don't know if it is a problem but:

While in the process of updating the "V2lRos2Publisher.cs", when I tried to read the topic from the terminal:
`~$ ros2 topic echo /perception/traffic_light_recognition/external/traffic_signals`
`The message type 'autoware_perception_msgs/msg/TrafficSignalArray' is invalid`
 
 I've checked it with the binary from current main and got the same result.
 I'm not really familiar with the Ros2 environment, so wanted to ask if it is the expected behaviour or not.

lchojnack, 2024-06-06:
@mozhoku 
I've encountered the same issue during migration for [AWSIM](https://github.com/tier4/AWSIM/), and it is definitely not expected. 

From the AWSIM side regarding V2I, I've applied the following commits https://github.com/tier4/AWSIM/commit/47f8df8905b43add7e251b8f61c3ccf3d0a32a62 and https://github.com/tier4/AWSIM/commit/b5e7c58c4999432b7781b123b2df4de9a57c130e
From the Autoware side, I've just build the latest main.

The issue not appear anymore. 

I think that could be caused by some `autoware_msgs` changes like e.g. https://github.com/autowarefoundation/autoware_msgs/commit/9de9a00b7034be2519c8b3257a4c4ee9e8eba087 (the change is quite old, and is strange that the issue not appeared during PR https://github.com/autowarefoundation/AWSIM-Labs/pull/76)

mozhoku, 2024-07-02:
New version is released here:
https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.1.0

closed 2024-07-02 by mozhoku