autowarefoundation/AWSIM-Labs #89: The vehicle cannot remain stable when returned to the starting point

meliketanrikulu, 2024-06-05:
I take the vehicle back to the starting point and wait for it to stand still. But it's starting to slip. The error does not occur every time, sometimes it works as expected. I tested with autoware 's sample vehicle.
You can find the test video showing the problem [here](https://www.youtube.com/watch?v=Xtd-zPTfDAs).

mozhoku, 2024-06-27:
@meliketanrikulu 
That might be caused by Autoware control. I've tried it in the editor with manual gears for the vehicle.
 When gear is in:
 Parking: It does not slip.
 Drive: little sliding.
 Neutral & Reverse: it just slides back with no resistance.
 
 Video: https://youtu.be/a7cRKTTqCyY

closed 2024-06-27 by mozhoku

WJaworskiRobotec, 2024-07-03:
Proper way of resetting position is following :