autowarefoundation/AWSIM-Labs #95: ROS2 not working

ceccocats, 2024-07-01:
Hi,
I just cloned main and have this error with ROS library:
```
RuntimeError: failed to get symbol 'rmw_init_options_init' due to Environment variable 'AMENT_PREFIX_PATH' is not set or empty, at ./src/functions.cpp:171, at ./src/rcl/init_options.c:75
ROS2.Utils.CheckReturnEnum (System.Int32 ret) (at <2034886caef046439cfa6ebcd9ed8cd7>:0)
ROS2.Ros2cs.Init () (at <2034886caef046439cfa6ebcd9ed8cd7>:0)
ROS2.ROS2ForUnity..ctor () (at Assets/Ros2ForUnity/Scripts/ROS2ForUnity.cs:326)
ROS2.ROS2UnityCore..ctor () (at Assets/Ros2ForUnity/Scripts/ROS2UnityCore.cs:55)
AWSIM.SimulatorROS2Node.Initialize () (at Assets/AWSIM/Scripts/ROS/SimulatorROS2Node.cs:37)
This error is the first then each ROS commands fails.
Do you know what could be the issue? do i need to set AMENT_PREFIX_PATH?
```
I am already using original version of AWSIM without any issue.

xmfcx, 2024-07-02:
Could you try the https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.1.0 release first and tell us if it works for you?

mozhoku, 2024-08-15:
@ceccocats Were you able to resolve the issue? What's the status?

xmfcx, 2024-09-03:
Closing this since no response from @ceccocats .

Please re-open when you are able to provide more information.

Juanpalai, 2024-11-06:
Hi,

I am also facing the same issue with version 1.1.0. It seems like I am unable to connect to anything related to ROS2. I am getting the following errors:

This is only with the Odometry Sensor, but I have a similar error with the other sensors that use ROS.
```
NullReferenceException: Object reference not set to an instance of an object
AWSIM.SimulatorROS2Node.UpdateROSTimestamp (ROS2.MessageWithHeader& message) (at Assets/AWSIM/Scripts/ROS/SimulatorROS2Node.cs:95)
AWSIM.OdometryRos2Publisher.Publish (AWSIM.OdometrySensor+OutputData outputData) (at Assets/AWSIM/Scripts/Sensors/Odometry/OdometryRos2Publisher.cs:105)
AWSIM.OdometrySensor.FixedUpdate () (at Assets/AWSIM/Scripts/Sensors/Odometry/OdometrySensor.cs:103)
```

And I’m also seeing the following error related to `rmw_init_options_init`:

```
RGLException: failed to get symbol 'rmw_init_options_init' due to Environment variable 'AMENT_PREFIX_PATH' is not set or empty, at ./src/functions.cpp:171
RGLUnityPlugin.RGLNativeAPI.CheckErr (System.Int32 status) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:244)
RGLUnityPlugin.RGLNativeAPI.NodePointsRos2PublishWithQos (System.IntPtr& node, System.String topicName, System.String frameId, RGLUnityPlugin.RGLQosPolicyReliability qos_reliability, RGLUnityPlugin.RGLQosPolicyDurability qos_durability, RGLUnityPlugin.RGLQosPolicyHistory qos_history, System.Int32 qos_depth) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:467)
RGLUnityPlugin.RGLNodeSequence.AddNodePointsRos2Publish (System.String identifier, System.String topicName, System.String frameId, RGLUnityPlugin.RGLQosPolicyReliability reliability, RGLUnityPlugin.RGLQosPolicyDurability durability, RGLUnityPlugin.RGLQosPolicyHistory history, System.Int32 historyDepth) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNodeSequence.cs:222)
AWSIM.RglLidarPublisher.OnEnable () (at Assets/AWSIM/Scripts/Sensors/LiDAR/RglLidarPublisher.cs:60)
```

xmfcx, 2024-11-06:
@Juanpalai Right now the latest version is: https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.5.3
Could you try that one please?

Additionally, the ROS 2 should not be sourced at the terminal that you are trying to launch AWSIM Labs in.

So could you open up a terminal and follow these steps:

```
# This should return empty
echo $AMENT_PREFIX_PATH

# This should return empty
echo $ROS_DISTRO

/path/to/awsim_labs_v1.5.3/awsim_labs.x86_64
```

Could you try this? Let us know if the echos return empty or not.

Juanpalai, 2024-11-06:
Just to clarify, I’m not using the demo; I’m trying to open AWSIM Labs directly from Unity. I also encountered an issue with Unity version 2022.3.36f1, which is the version mentioned as being compatible with AWSIM Labs v1.5.3. Unity keeps crashing either when loading the project or when trying to export the packages. As a workaround, I switched to Unity version 2022.3.50f1.

I’ve already tried AWSIM Labs v1.5.3, but the ROS2 problem persists. Could this Unity version mismatch be causing the issue, or do you think it’s unrelated?

xmfcx, 2024-11-06:
## Test with the demo first

I understand. But to help us debug the problem, can you first run the demo and report if it works with the instructions that I've provided above? Without unity.

## Unity crashing issue

We've been using `2022.3.36f1` in multiple systems for a while and did not experience such crashes.

But we know that an incompatible NVIDIA driver version will cause these random crashes.

The compatible NVIDIA Driver version is: `545.23.08`

Could you check it with the following command?
```console
$ nvidia-smi
Wed Nov  6 12:06:49 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.23.08              Driver Version: 545.23.08    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
```

## The actual problem

Similarly, the unityhub should also be started in an environment without ROS 2 being sourced.

To confirm,

1. Quit unityhub from the system tray (this is important).
2. Open up a terminal.
3. Run `echo $ROS_DISTRO` and it should return empty. Please tell us if it returns `humble`
4. Type `unityhub` in the terminal and press Enter.
5. Open up the project and let us know if the problem still persists.

Juanpalai, 2024-11-07:
I’ve tried running the demo, and it works without any issues.

Regarding my system configuration, my current NVIDIA driver version is 555.42.06:

```shell
$ nvidia-smi
Thu Nov  7 11:09:29 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 555.42.06              Driver Version: 555.42.06      CUDA Version: 12.5     |
|-----------------------------------------+------------------------+----------------------+
```

However, Unity version 2022.3.36f1 keeps crashing on my setup. Should I consider downgrading the driver to the compatible 545.23.08 version?

In any case, I managed to resolve the ROS2 issue. It turned out that I had ROS2 sourced in my `~/.bashrc`. After removing that line, following the DDS configuration instructions from the documentation, and adhering to the steps you provided, everything is now working smoothly. 

Thank you for the guidance!

xmfcx, 2024-11-07:
I'm glad you've solved your problem! <3

For CUDA/Driver compatibilityi please see the official docs: https://docs.nvidia.com/deploy/cuda-compatibility/#id3

Here, you will see that CUDA 12.3 (the version installed with and supported by Autoware) is only compatible with `545.23.06+`.

In order to work with Unity under Ubuntu, I would definitely do that.
- Always be careful when changing NVIDIA Driver versions, they can be very finicky.

By the way, if the updated Unity version is not crashing even with your more up-to-date driver version, you may also consider keeping it and see how things go. :+1: 

I'm closing this issue since it is resolved, feel free to open up a new issue if the need arises.

closed 2024-11-07 by xmfcx