autowarefoundation/AWSIM-Labs #104: Toggling lidar sensors causes fps drop

mozhoku, 2024-07-08:
Toggling lidar sensors causing fps drop in the simulator. It is not consistent with it.

I also had problem with removing publisher topics after destroying the related component. After looking a bit more into the code I saw a part saying that GC should take care of it by default. It was inconsistent with removal of the dead topics. So I had to call GC manually after disabling the sensors. I'm suspecting a similar issue with the lidars but I'm not sure. I tried calling GC after removal of the RGLLidarPublisher but couldn't see any difference. 

Since lidar simulation works different from the other sensors in the simulation, I'm kinda stuck here. I'm not really experienced about GPU side of the stuff so any help would be appreciated :pray: 

Here is the video showing the issue:
- https://youtu.be/OguyhREBRbk

Script responsible for toggling the sensors (part related to lidars in lines: 16-33): 
- https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/AWSIM/Scripts/UI/SensorToggleFunctions.cs

Current RGLLidarPublisher and LidarSensor scripts:
- https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/AWSIM/Scripts/Sensors/LiDAR/RglLidarPublisher.cs
- https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/RGLUnityPlugin/Scripts/LidarSensor.cs

@WJaworskiRobotec @lchojnack

msz-rai, 2024-07-08:
Hi @mozhoku 

Thank you for reporting the issue.

The FPS drop when toggling LiDARs may be due to their desynchronization and the need to update the scene to RGL (library for LiDAR simulation) more frequently. This update is an expensive operation, especially when skinned meshes are on the scene (mesh skinning is performed on the CPU). When starting the simulation, the timers of the LiDARs are initialized identically. On each `FixedUpdate()` the timer is incremented with `Time.deltaTime` and the check whether to trigger the sensor is performed. Before each LiDAR trigger `SceneManager` updates the scene to RGL. If all of the LiDARs trigger on the same "frame" there is only one scene update needed (the best case for FPS).

When LiDAR is disabled its timer is not updated. After its activation, its timer will not be synchronized with other LiDARs (unless activated at the right moment, but that is unlikely) and there will be additional scene updates needed.

If my assumptions are correct, the quick workaround would be to assign the [timer value](https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/RGLUnityPlugin/Scripts/LidarSensor.cs#L93) to the timer of any other active sensor in the [OnEnable()](https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/RGLUnityPlugin/Scripts/LidarSensor.cs#L218) method.

msz-rai, 2024-07-08:
@mozhoku 

Regarding removing publisher topics, the code in [OnDestroy()](https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/AWSIM/Scripts/Sensors/LiDAR/RglLidarPublisher.cs#L169) method should destroy publishers immediately.

mozhoku, 2024-07-09:
@msz-rai 
Thank you for the reply :pray: I'll try to keep their timer values in sync. 

> @mozhoku
> 
> Regarding removing publisher topics, the code in [OnDestroy()](https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/AWSIM/Scripts/Sensors/LiDAR/RglLidarPublisher.cs#L169) method should destroy publishers immediately.

This is true for the lidar publishers, they remove their topics OnDestroy() immediately. But for other sensors I have to call GC manually.

It says in the: https://github.com/autowarefoundation/AWSIM-Labs/blob/6c37af514407c3192b20a47a1c610fe323496daa/Assets/Ros2ForUnity/Scripts/ROS2Node.cs#L23C4-L27C19
    
    /// A class representing a ros2 node. Multiple nodes can be used. Node can be removed by GC when not used anymore,
    /// but will also be removed properly with Ros2cs Shutdown, which ROS2 for Unity performs on application quit
    /// The node should be constructed through ROS2UnityComponent class, which also handles spinning

mozhoku, 2024-07-10:
It was just as you said @msz-rai . Thanks a lot :+1: 

Here is the video with synced lidar timers: https://youtu.be/NL3mylAa7zI

closed 2024-07-10 by mozhoku