autowarefoundation/AWSIM-Labs #108: Vehicle Physics Pro Community Edition Integration

mozhoku, 2024-07-10:
Default tools in Unity is simple and can't simulate vehicle behaviour properly. Using VPP for more detailed vehicle simulation would be better for the AWSIM Labs users. It also can handle different types of [vehicle configurations](https://vehiclephysics.com/about/features/#feature-examples).

For further information about the package:
Asset Store: https://assetstore.unity.com/packages/tools/physics/vehicle-physics-pro-community-edition-153556
Website: https://vehiclephysics.com/

sasa42, 2024-07-11:
+1

fyi,  fantastic project done with VPP...

https://github.com/PERRINN/project-424-unity

I did an HDRP test on an older version.

https://github.com/sasa42/project-424-unity.git
branch HDRP_Drive_Test
https://youtu.be/EyVYKxtS6uE?feature=shared

sasa42, 2024-07-17:
I think Edy Garcia would provide an integration script so that VPP switch works in one click. I will give him a message and link to this issue.

closed 2024-09-13 by mozhoku

roboserg, 2025-05-17:
Any updates on this?

xmfcx, 2025-05-17:
VPP has been integrated and being used within AWSIM Labs ✅