autowarefoundation/AWSIM-Labs #110: Traffic simulation is struggling to simulate when there are lots of NPC's

mozhoku, 2024-07-10:
Simulation slows down when there are lots of NPC's in the scene.

WJaworskiRobotec, 2024-07-17:
Please specify target number of NPCs @mozhoku that we would like to support and test

sasa42, 2024-07-17:
I really like this old ecs example for visual traffic and pedestrians.
https://github.com/maajor/OSMTrafficSim

Probably it could be useful to solve some general performance issues. 
On my PC i can simulate around 50 but the framerate drops to 8fps.  
The ecs example can run 10 000+ at high framerates.

lchojnack, 2024-09-10:
I've started to work on this task. I'm exploring the posibility to use the [ECS (Entity Component System)](https://unity.com/ecs) for traffic simulation.

lchojnack, 2024-12-17:
The current task progress:
- Basic ECS Traffic Simulator is ready -> NPCs drive on lanelet
  - There is around 80 FPS for 1000 NPCs 

https://github.com/user-attachments/assets/1ac41ab5-cd60-47f2-ac6c-6c99199f924b

lchojnack, 2025-02-19:
The current task progress:

- NPC collision avoidance in ECS Traffic Simulator initial implementation
  - NPC can stop in front of EGO, and resume to drive when EGO disapear.
  
    https://github.com/user-attachments/assets/a7f1ff9b-120a-4528-a629-27453dfaca6c

lchojnack, 2025-04-16:
The current task progress:
- add NPC right of way initial implementation in ECS Traffic Simulator

  https://github.com/user-attachments/assets/dda5279a-0920-4557-bf2d-da5aa4beb8a8