autowarefoundation/AWSIM-Labs #115: Lidar simulation is not working in WSL2

xmfcx, 2024-07-11:
...  some lidar are missing....
and the pointcloud_raw_ex

[image]

_Originally posted by @sasa42 in https://github.com/autowarefoundation/AWSIM-Labs/issues/87#issuecomment-2222514692_

xmfcx, 2024-07-11:
@msz-rai do you have any experience running the https://github.com/RobotecAI/RobotecGPULidar on WSL2?

xmfcx, 2024-07-11:
@sasa42 could you also share the nvidia-smi results with us?

[Someone said:](https://forums.developer.nvidia.com/t/enabling-nvidia-support-for-vulkan-on-ubuntu-22-04-through-wsl2/244176)
> don’t install the nvidia driver, there’s no real nvidia gpu available in WSL2.

my output:

```console
mfc@mfc-leo:~$ nvidia-smi
Thu Jul 11 13:32:27 2024       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 545.29.06              Driver Version: 545.29.06    CUDA Version: 12.3     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3080        Off | 00000000:2D:00.0  On |                  N/A |
| 53%   75C    P2             148W / 320W |   3447MiB / 10240MiB |     23%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1583      G   /usr/lib/xorg/Xorg                          838MiB |
|    0   N/A  N/A      5867      G   ...yOnDemand --variations-seed-version       42MiB |
|    0   N/A  N/A      6345      G   ...,262144 --variations-seed-version=1      226MiB |
|    0   N/A  N/A   1131802      G   ...erProcess --variations-seed-version       23MiB |
|    0   N/A  N/A   1164733      G   ...erProcess --variations-seed-version       65MiB |
|    0   N/A  N/A   2589250    C+G   ./awsim_labs_v1.1.1/awsim_labs.x86_64      2198MiB |
|    0   N/A  N/A   3169233      G   ...seed-version=20240708-050118.992000       15MiB |
+---------------------------------------------------------------------------------------+
```

sasa42, 2024-07-11:
this is the Unity PlayerLog....

[image]
[image]

mozhoku, 2024-07-11:
@sasa42 You shouldn't source ros2 when launching AWSIM.

sasa42, 2024-07-11:
...uh. 
What dou yo mean with source ros2.

I thsi case i only started  ...v1.1.1/
./awsim_labs.x86_64

I deleted the log files several times and redid the awsim start.

mozhoku, 2024-07-11:
The errors you've posted says that you've sourced Ros2. 
Please check your .bashrc file for this line:
`source /opt/ros/humble/setup.bash` and comment it out.
What this line does is when you open a fresh terminal it sources ros2 right away. With this disabled you will have to do it manually when you need it.

Also please launch the awsim from a fresh separate terminal without sourcing the ros2.

xmfcx, 2024-07-11:
.bashrc file is located at the home directory.

you can open it with:
```
sudo apt install gedit
gedit ~/.bashrc
```

And modify the file.

xmfcx, 2024-07-11:
But you have other Nvidia Optix related errors, this still won't work.

Your only option is to dual boot with Ubuntu, run both AWSIM Labs and Autoware in native ubuntu 22.04

sasa42, 2024-07-11:
... this is the log after  #source....
and restart in a new terminal.

[image]

sasa42, 2024-07-11:
needed configs are...

Autoware WSL - AWSIM WIN 1PC
Autoware WSL - AWSIM WSL 1PC

Autoware WSL - AWSIM WIN 2PC
Autoware Native - AWSIM WIN 2PC
Autoware Native - AWSIM MAC 2PC
in this order... .)

xmfcx, 2024-07-11:
For me it says:
```log
RGL mesh source: RegularMeshesAndSkinnedMeshes
[Thu Jul 11 14:47:59 2024]: Logging configured: level=info, file=(disabled), stdout=true
[14:47:59][    13 us][info]: RGL Version 0.16.2 branch=develop commitSHA1=8eee9475ced107603063533e6333498252dd0ba8
[14:47:59][ 18572 us][info]: Running on GPU: NVIDIA GeForce RTX 3080
[14:47:59][    23 us][info]: Built against OptiX SDK version: 7.2.0
[14:47:59][     4 us][info]: Built against OptiX ABI version: 41
[14:47:59][     4 us][info]: Built against CUDA Toolkit version: 11.7
[14:47:59][     9 us][info]: Installed CUDA runtime version: 11.7
[14:47:59][     4 us][info]: Installed CUDA driver version: 12.3
[14:47:59][ 89019 us][info]: Installed NVidia kernel driver version: 545.29.06
```

Full `/home/mfc/.config/unity3d/AWF/AWSIM Labs/Player.log`: https://gist.github.com/xmfcx/a1e70ac4e0c00f980f5c207a64726ab2

AWSIM Labs or AWSIM probably won't run on WSL2. @lchojnack @WJaworskiRobotec @msz-rai are from Robotec and their team developed the Lidar Unity simulator. Maybe they can help but seeing awsim also slow, I wouldn't recommend it.

prybicki, 2024-07-11:
@xmfcx By looking on the errors, my quick intuition is that either CUDA driver support is not full on WSL2, or implicit initialization behavior of the CUDA driver is different on WSL2 than everywhere else and it doesn't get automatically initialized before constructing OptiX context. If you're willing to recompile RGL and debug, the relevant place to look is [here](https://github.com/RobotecAI/RobotecGPULidar/blob/develop/src/Optix.cpp#L50). 

> Maybe they can help but seeing awsim also slow, I wouldn't recommend it.

Would you mind expanding your thought?

sasa42, 2024-07-11:
Thanks for the gist to compare.
I have to solve it so i will do an deepdive.

sasa42, 2024-07-11:
There are some test on 4 pages from last year....
https://www.phoronix.com/review/windows11-wsl2-zen4/2

the WSL2bs native  performance doesn t look so bad.

sasa42, 2024-07-11:
> @sasa42 could you also share the `nvidia-smi` results with us?
> 
> [Someone said:](https://forums.developer.nvidia.com/t/enabling-nvidia-support-for-vulkan-on-ubuntu-22-04-through-wsl2/244176)
> 
> > don’t install the nvidia driver, there’s no real nvidia gpu available in WSL2.
> 
> my output:
> 
> ```
> mfc@mfc-leo:~$ nvidia-smi
> Thu Jul 11 13:32:27 2024       
> +---------------------------------------------------------------------------------------+
> | NVIDIA-SMI 545.29.06              Driver Version: 545.29.06    CUDA Version: 12.3     |
> |-----------------------------------------+----------------------+----------------------+
> | GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
> | Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
> |                                         |                      |               MIG M. |
> |=========================================+======================+======================|
> |   0  NVIDIA GeForce RTX 3080        Off | 00000000:2D:00.0  On |                  N/A |
> | 53%   75C    P2             148W / 320W |   3447MiB / 10240MiB |     23%      Default |
> |                                         |                      |                  N/A |
> +-----------------------------------------+----------------------+----------------------+
>                                                                                          
> +---------------------------------------------------------------------------------------+
> | Processes:                                                                            |
> |  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
> |        ID   ID                                                             Usage      |
> |=======================================================================================|
> |    0   N/A  N/A      1583      G   /usr/lib/xorg/Xorg                          838MiB |
> |    0   N/A  N/A      5867      G   ...yOnDemand --variations-seed-version       42MiB |
> |    0   N/A  N/A      6345      G   ...,262144 --variations-seed-version=1      226MiB |
> |    0   N/A  N/A   1131802      G   ...erProcess --variations-seed-version       23MiB |
> |    0   N/A  N/A   1164733      G   ...erProcess --variations-seed-version       65MiB |
> |    0   N/A  N/A   2589250    C+G   ./awsim_labs_v1.1.1/awsim_labs.x86_64      2198MiB |
> |    0   N/A  N/A   3169233      G   ...seed-version=20240708-050118.992000       15MiB |
> +---------------------------------------------------------------------------------------+
> ```

On the left i have some detailed debug output...

(https://github.com/autowarefoundation/AWSIM-Labs/assets/1211326/1b06fdc4-380a-4861-8048-593c5baab9c2)

Do you need the nvidia-smi, too

xmfcx, 2024-07-11:
@prybicki 
>> Maybe they can help but seeing awsim also slow, I wouldn't recommend it.

> Would you mind expanding your thought?

He has a pc with high end parts: 7945HX cpu and rtx 4090 yet he has very low framerates.
I've assumed this was due to running in WSL2, and since it is running slow, maybe it will keep running slow even if lidar simulation is fixed. So this is why I don't recommend running AWSIM in WSL2.

Maybe we should fix the problem of AWSIM Labs not running on windows.

Then he could run autoware on wsl2 and awsim labs on native windows.

sasa42, 2024-07-11:
nvidia-smi...

[image]

xmfcx, 2024-09-03:
Need to add that we don't support WSL to the documentation.

mozhoku, 2024-09-27:
We will not be supporting the WSL2, updated the documentation. Closing the issue.

closed 2024-09-27 by mozhoku