autowarefoundation/AWSIM-Labs #117: Incomplete guidance on OpenScenario in documentation

Aysenayilmaz, 2024-07-12:
The current section on using OpenScenario in the documentation does not provide sufficient guidance for new users. Specifically, there is no section on how to make a .yaml file created with the Scenario Editor usable in Awsim Labs. Additionally, the branch name and sensor kit name required for Autoware are incorrect.

WilliamFlinchbaugh, 2024-07-17:
Is there anyone who can confirm if scenario_simulator_v2 actually works with AWSIM-Labs? I've been having trouble even getting it to work with regular AWSIM, and there doesn't seem to be an AWSIM-Labs ss2 package. I haven't been able to try through the Unity editor yet due to some issues with Unity crashing.

The documentation on OpenSCENARIO seems like it's essentially unchanged from the regular AWSIM docs. Would be nice to have it updated.

Aysenayilmaz, 2024-07-18:
@WilliamFlinchbaugh Currently, there is no issue with using ss2 with AWsim-Labs.

- https://www.youtube.com/watch?v=hfdOi4xIQqM 

We will update the documentation also add a new binary file and the edits required for using the created scenarios in AWsim-Labs.

Aysenayilmaz, 2024-07-23:
Changes are merged. 

-  https://github.com/autowarefoundation/AWSIM-Labs/pull/120