autowarefoundation/AWSIM-Labs #119: AWSIM Labs executable not launching properly in Windows

mozhoku, 2024-07-17:
- When trying to start simulation from the editor, it also gets stuck on play or it freezes the editor. 
- The executable gets stuck at 200mb~ ram usage and doesn't start.  

This problem occurs in my PC while it works without any problem in other computers at home. And it only happens in awsimLabs, my other Unity projects are unaffected.
 
@sasa42 shares the same issue with me.

Things I've tried with @xmfcx :
- Update / Downgrade the graphics card drivers. (with a working machine)
- Changed Unity Editor version with awsimLabs.
- Freshly cloned repository,
- Fresh Unity Editor installations,

I'll share more details later on.

sasa42, 2024-07-17:
Hi. Since yesterday i have the same behaviour.

mentioned here:
https://autowarefoundation.github.io/AWSIM-Labs/pr-90/DeveloperGuide/TroubleShooting/

(Windows) Built-binary or Unity Editor freeze when simulation started

Unity project play mode and starting of the build results in freeze....

i also:

- updated / downgraded the graphics card drivers
- updated / downgraded Network Interface Drivers ( reference to Trouböeshooting solution)
- recloned and rebuild awsim and awsim labs
- disabled/enabled firewall

The Unity projects and builds worked for me the last 3 weeks on daily use.
This behaviour strated yesterday on my side.

Other Unity projects and builds are not effected. 
Only all old awsim and  new awsim labs build and projects.
When i copy this .exe to another PC it works again.

mozhoku, 2024-08-26:
I had to format my PC last week. Didn't get this problem with the new installation yet.