autowarefoundation/AWSIM-Labs #121: Parking Planner Requirements

armaganarsln, 2024-07-18:
In order to test of new parking planner in this https://github.com/orgs/autowarefoundation/discussions/4936

We need a simulation testing environment. The environment should consist of different types of parking areas such as regular, angled, and parallel parking. Here is an example image of how it could be.

The parking areas for parallel parking will be 2.5m wide and the length will be 1.5 x the car length. The length will depend on the selected vehicle.

The road in the area should be two-way in all the parking areas and have a 7m width.

[image]

For checking a similar test here is a video done by a Tesla driver: https://www.youtube.com/watch?v=2dbPrltSTp0

In the simulation environment, the parking areas should be busy and vehicles should be trying to park between cars in most test cases.

xmfcx, 2024-07-18:
We can start this once:
- https://github.com/autowarefoundation/AWSIM-Labs/pull/96
is merged.

sasa42, 2024-07-18:
I like to put this map into the loop...

[image]

[image]

Zala Zone Smart City

I did a quick review because i searched a test ground with complete city (focused on an  adas feature set)

I downloaded the Unity Road and Env mesh. Also the point clouds and found a LanLet2 set for smart city on github.

So there would me a full set of scenarios and even a big parking place. ( right on sscreenshot)

I will add some shots from the data later.

sasa42, 2024-07-19:
Mesh....
1+2+3 could be extended parking spaces

[image]

OSM Lanes
[image]

LaneLet2
https://github.com/jkk-research/lanelet2_maps

armaganarsln, 2024-07-22:
@sasa42 after preparing the required map we will try to import this one as well and test. Our aim is to create a very simple one so that the developers in planning could test with their laptop. Currently they are using 2D simulator and we are trying to move them to AWSIM. This area might be too big and might result in performance decrease therefore I can't promise but we will definitely test and give you feedback. Thank you for your suggestion.

Aysenayilmaz, 2024-07-25:
**_Update on the current status of the environment:_**

Three different parking area were created: parallel parking, L parking, and angled parking. Adjustments and improvements will be made to the environment. It will also be tested and reported using the current parking planner.

[image]
[image]
[image]
[image]

### **Dimensions of the Parking Spaces**
_(The unit of measurement is in metres.)_

- Parallel Parking: 3 x 7.275 (The dimensions were set to 1.5 times the size of the Lexus vehicle.)
[image]

- L Parking: 2.5 x 6
[image]

- Angled parking : 7 x 3 & 60-degree
[image]

Aysenayilmaz, 2024-08-01:
**_Update on the current status of the environment:_** 
- The pointcloud map and HD-map of the environment have been created.
- L parking has been tested.
https://youtu.be/7PW41OYb4b8

- Parallel parking has been tested.
https://youtu.be/A4k5VDHK4eg

- Angled parking has been tested.
https://youtu.be/RS-P24wmZxE

Aysenayilmaz, 2024-08-01:
The binary file, pcd, and HD map have been added to the new release. https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.2.0 

- The binary file :  [awsim_labs_parking_v1.2.0.zip](https://github.com/autowarefoundation/AWSIM-Labs/releases/download/v1.2.0/awsim_labs_parking_v1.2.0.zip)  
- pcd and hd-map :  [awsim_labs_ParkinArea.zip](https://github.com/autowarefoundation/AWSIM-Labs/releases/download/v1.2.0/awsim_labs_ParkinArea.zip)

armaganarsln, 2024-08-02:
Thanks @Aysenayilmaz , since the binaries are released, we can close this issue.

closed 2024-08-02 by armaganarsln

Aysenayilmaz, 2024-08-13:
The dimensions of the parking spaces were made based on this reference: 

- https://cdn-wordpress.webspec.cloud/intrans.iastate.edu/uploads/sites/15/2018/12/Chapter_08-2018.pdf
- https://www.dimensions.com/element/30-degree-parking-spaces-layouts