autowarefoundation/AWSIM-Labs #125: Null references and no ROS topics when running simulation from Unity Editor

WilliamFlinchbaugh, 2024-07-25:
I'm getting tons of null references and no topics when running either the AutowareSimulation or AutowareSimulationScenario scenes. The regular demo and ss2 demo work fine.

What I've tried:
* I setup my DDS configuration correctly (However I don't have ROS_LOCALHOST_ONLY=1 as per the docs instructions)
* I don't have ROS2 sourced in .bashrc or .profile, so the editor should not have ROS2 sourced
* Tried cloning the repository again
* Tried with both Nishishinjuku_URP_v0.1.0 and v0.2

After running the simulation from the editor, calling `ros2 topic list` only returns `/parameter_events` and `/rosout`.

Note: I downgraded my nvidia drivers to 535 because the editor would crash on startup with 555 and 550.

[image]

Aysenayilmaz, 2024-07-26:
@WilliamFlinchbaugh Are you running UnityHub from the terminal? You can get the same error if you open the editor and start the scene without running UnityHub from the terminal. 
[image]

WilliamFlinchbaugh, 2024-07-28:
@Aysenayilmaz Yep, that was it. I missed the little warning in the docs. This might be worth putting in the troubleshooting section as well, but maybe I'm just stupid. Thank you!

closed 2024-07-28 by WilliamFlinchbaugh