autowarefoundation/AWSIM-Labs #126: Unable to initialize pose only when running in Unity editor

WilliamFlinchbaugh, 2024-07-30:
I'm currently running into an issue where autoware is unable to initialize the pose, but only when running AWSIM Labs in the unity editor. I have no issues when I use the v1.1.1 package. This occurs using the regular simulation scene and the scenario simulation scene. 

I'm running on the awsim-labs-stable branch of autoware. I don't get any errors or warnings within Unity, and there seems to be no issue connecting since I can set the pose in autoware and it will teleport the car. The ROS topic list looks correct to me as well.

In autoware, it shows that the vehicle is in motion, and the console output looks like this and keeps repeating:

```
[pose_initializer_node-38] [INFO] [1722314316.215694186] [localization.util.pose_initializer]: EKF triggering service is available!
[pose_initializer_node-38] [INFO] [1722314316.215944834] [localization.util.pose_initializer]: EKF Deactivation succeeded
[pose_initializer_node-38] [INFO] [1722314316.216031964] [localization.util.pose_initializer]: NDT triggering service is available!
[pose_initializer_node-38] [INFO] [1722314316.216294572] [localization.util.pose_initializer]: NDT Deactivation succeeded
[pose_initializer_node-38] [INFO] [1722314316.216317262] [localization.util.pose_initializer]: fit_target: pointcloud_map, frame: map
[ekf_localizer_node-40] [WARN] [1722314316.874195032] [localization.pose_twist_fusion_filter.ekf_localizer]: The node is not activated. Provide initial pose to pose_initializer
[component_container_mt-62] [INFO] [1722314316.888356272] [planning.mission_planning.mission_planner]: waiting odometry... Route API is not ready.
[pose_initializer_node-38] [INFO] [1722314316.999857242] [localization.util.pose_initializer]: Call NDT align server.
[ndt_scan_matcher_node-36] [WARN] [1722314317.000599897] [localization.pose_estimator.ndt_scan_matcher]: No InputSource. Please check the input lidar topic
[service_log_checker_node-4] [ERROR] [1722314317.000933245] [system.service_log_checker]: /localization/initialize: status code 4 'NDT align server failed.' (/localization/util/pose_initializer)
[service_log_checker_node-4] [ERROR] [1722314317.001333952] [system.service_log_checker]: /localization/initialize: status code 4 'NDT align server failed.' (/default_ad_api/node/localization)
[service_log_checker_node-4] [ERROR] [1722314317.001355852] [system.service_log_checker]: /api/localization/initialize: status code 4 'NDT align server failed.' (/default_ad_api/node/localization)
[service_log_checker_node-4] [ERROR] [1722314317.001507930] [system.service_log_checker]: /api/localization/initialize: status code 4 'NDT align server failed.' (/localization/util/default_ad_api/helpers/automatic_pose_initializer)
[map_based_prediction-53] [WARN] [1722314317.078401516] [perception.object_recognition.prediction.map_based_prediction]: failed to get transform from base_link to map: Could not find a connection between 'base_link' and 'map' because they are not part of the same tree.Tf has two or more unconnected trees.
[topic_state_monitor_node-20] [INFO] [1722314317.479131998] [system.topic_state_monitor_transform_map_to_base_link]: /tf has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-11] [INFO] [1722314317.479151487] [system.topic_state_monitor_object_recognition_objects]: /perception/object_recognition/objects has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-10] [INFO] [1722314317.479198067] [system.topic_state_monitor_obstacle_segmentation_pointcloud]: /perception/obstacle_segmentation/pointcloud has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-9] [INFO] [1722314317.479244477] [system.topic_state_monitor_pose_twist_fusion_filter_pose]: /localization/pose_twist_fusion_filter/pose has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-16] [INFO] [1722314317.479318766] [system.topic_state_monitor_control_command_control_cmd]: /control/command/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-8] [INFO] [1722314317.479493655] [system.topic_state_monitor_initialpose3d]: /initialpose3d has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-15] [INFO] [1722314317.479555895] [system.topic_state_monitor_trajectory_follower_control_cmd]: /control/trajectory_follower/control_cmd has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-14] [INFO] [1722314317.480092171] [system.topic_state_monitor_scenario_planning_trajectory]: /planning/scenario_planning/trajectory has not received. Set ERROR in diagnostics.
[topic_state_monitor_node-13] [INFO] [1722314317.480357399] [system.topic_state_monitor_mission_planning_route]: /planning/mission_planning/route has not received. Set ERROR in diagnostics.
```

mozhoku, 2024-07-30:
Hi @WilliamFlinchbaugh,
If you are using the main branch of AWSIM Labs, it no longer works with the awsim-stable branch. It should work with the autoware-main branch. 

Release v1.1.1 works with awsim-stable branch. 

There was a change in the pointcloud message structure in the Autoware side. We've created a stable branch to AWSIM Labs to keep it working with Autoware as a temporary solution. We haven't released new version after this change yet. I'll comment here when we update the release version. :+1:

WilliamFlinchbaugh, 2024-07-30:
Seems to work, thank you!