autowarefoundation/AWSIM-Labs #138: AWSIM keyboard input not working

mitsudome-r, 2024-09-03:
I have noticed that there is a guide about VehicleKeyboardInput.cs shown in the Hotkeys panel, but vehicle does not respond to the keyboard input.

We should either remove the guide if the function is not on, or we should enable the function by default.

[image]

Aysenayilmaz, 2024-09-03:
@mitsudome-r  Before driving with the keyboard, you need to click on the keyboard icon under the Ego Vehicle section. Did you do this? 

[image]

mitsudome-r, 2024-09-05:
Thanks! I wasn't aware that there was a button there.

closed 2024-09-05 by mitsudome-r