autowarefoundation/AWSIM-Labs #157: Unable to spawn dummy car in AWSIM with e2e_simulator due to topic type conflict

batuhanbeytekin, 2024-09-24:
### Description
I'm currently using AWSIM v1.2.1 with e2e_simulator and encountered an issue when trying to spawn a dummy car using the 2D Dummy Car button. The car does not spawn, and when I try to echo the topic /simulation/dummy_perception_publisher/object_info, I receive the following error:

```
Cannot echo topic '/simulation/dummy_perception_publisher/object_info', as it contains more than one type: [dummy_perception_publisher/msg/Object, tier4_simulation_msgs/msg/DummyObject]
```
When I run ```ros2 topic info -v``` on this topic, I get the following output:
```
Type: ['dummy_perception_publisher/msg/Object', 'tier4_simulation_msgs/msg/DummyObject']

Publisher count: 4

Node name: rviz2
Node namespace: /
Topic type: tier4_simulation_msgs/msg/DummyObject
Endpoint type: PUBLISHER
GID: 01.10.f1.8c.84.79.da.0a.af.ef.d7.77.00.00.a0.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

...

Subscription count: 1

Node name: AWSIM
Node namespace: /
Topic type: dummy_perception_publisher/msg/Object
Endpoint type: SUBSCRIPTION
GID: 01.10.7c.19.e5.1e.30.1c.74.30.7c.61.00.00.0e.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```
It seems that there is a conflict between the message types being published to /simulation/dummy_perception_publisher/object_info, as both ```dummy_perception_publisher/msg/Object``` and ```tier4_simulation_msgs/msg/DummyObject``` are being used.
### Steps to reproduce
1. Download AWSIM v1.2.1 from [this link](https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.2.1) and extract the ```awsim_labs_v1.2.1.zip``` file.
2. Navigate to the extracted directory:
```cd path/to/awsim_labs_v1.2.1```

mozhoku, 2024-09-25:
I've tried to create same issue with the steps you've provided @batuhanbeytekin. NPC's can be spawned from the Autoware side without any problems:
```
rti -v /simulation/dummy_perception_publisher/object_info
Type: dummy_perception_publisher/msg/Object

Publisher count: 4

Node name: rviz2
Node namespace: /
Topic type: dummy_perception_publisher/msg/Object
Endpoint type: PUBLISHER
GID: 01.10.1b.bb.ba.5e.44.6e.92.52.ef.14.00.00.a1.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: rviz2
Node namespace: /
Topic type: dummy_perception_publisher/msg/Object
Endpoint type: PUBLISHER
GID: 01.10.1b.bb.ba.5e.44.6e.92.52.ef.14.00.00.a3.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: rviz2
Node namespace: /
Topic type: dummy_perception_publisher/msg/Object
Endpoint type: PUBLISHER
GID: 01.10.1b.bb.ba.5e.44.6e.92.52.ef.14.00.00.a5.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: rviz2
Node namespace: /
Topic type: dummy_perception_publisher/msg/Object
Endpoint type: PUBLISHER
GID: 01.10.1b.bb.ba.5e.44.6e.92.52.ef.14.00.00.a8.03.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Subscription count: 1

Node name: AWSIM
Node namespace: /
Topic type: dummy_perception_publisher/msg/Object
Endpoint type: SUBSCRIPTION
GID: 01.10.a0.51.3c.c9.97.20.d3.1d.ff.2c.00.00.0e.04.00.00.00.00.00.00.00.00
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (1)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

Aysenayilmaz, 2024-09-25:
When I tested with main Autoware and awsim_labs_v1.4.3, I encountered the same issue. The message type used for dummy vehicles has changed with the https://github.com/autowarefoundation/autoware.universe/issues/8704  PR merged into Autoware.universe, which is why we are facing this problem. The message types will be updated and fixed. 
Thank you for issue! @batuhanbeytekin

mozhoku, 2024-09-25:
It seems like my Autoware was not up to date, sorry :pray:

mozhoku, 2024-09-27:
@batuhanbeytekin Can you test this case with the latest release? It should be okay now.

batuhanbeytekin, 2024-09-27:
> @batuhanbeytekin Can you test this case with the latest release? It should be okay now.

I tested again with the latest version and it seems the problem is solved. Thanks for your effort.

closed 2024-09-27 by Aysenayilmaz