autowarefoundation/AWSIM-Labs #169: AWSIM Labs - Integration with Autoware not working

pandyah, 2024-09-30:
Simulator is not able to integrate with latest Autoware. It freezes in few seconds with localization error as below.

[image]

mozhoku, 2024-09-30:
Please try to launch the simulator from the terminal like this:

[image]

pandyah, 2024-09-30:
Thanks for the quick response. Yes, have not sourced ROS2. Have removed sourcing of ROS2 in .bashrc
Have tried starting as mentioned by you, but same issue.
I am running both autoware and AWSIM in the same laptop with Ubuntu 22.04. Not sure if that is causing the issue.
[
[image]
](url)

mozhoku, 2024-09-30:
AWSIM Labs and Autoware should be able to run with that configuration in the same PC. I'll check the current Autoware main to see if there are any issues :+1:

maxime-clem, 2024-09-30:
I found the following issue using the latest Autoware:
```
[component_container_mt-1] [WARN 1727692755.050491285] [sensing.lidar.concatenate_data] default_incompatible_qos_callback(): New subscription discovered on topic '/sensing/lidar/concatenated/pointcloud', requesting incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY:(L273)
```

mozhoku, 2024-09-30:
I was able to run AWSIM Labs v1.5.1 with up to date Autoware and Autoware Universe main. @pandyah Do you have anything different than default setup?

pandyah, 2024-10-01:
@mozhoku I have the AWSIM Labs v1.5.1  with latest (30th Sept) main branch of Autoware Universe.

mozhoku, 2024-10-01:
> @mozhoku I have the AWSIM Labs v1.5.1 with latest (30th Sept) main branch of Autoware Universe.

Okay have you followed these steps? 
- https://autowarefoundation.github.io/AWSIM-Labs/main/GettingStarted/QuickStartDemo/

There is some small differences from the Tier IV's AWSIM if you've used that one before.

pandyah, 2024-10-01:
@mozhoku Yes, followed the steps as mentioned in the link.

mozhoku, 2024-10-01:
@pandyah In that case there shouldn't be any problems from the AWSIM Labs side. 

Since I'm not very familiar with the Autoware, I'll try to ask my colleagues. Maybe they can help you out.

maxime-clem, 2024-10-01:
@pandyah Can you share the full Autoware log file ? You can find it in `$HOME/.ros/log/{DATE-...}/launch.log`

xmfcx, 2024-10-01:
@Aysenayilmaz this broke the Autoware side:
- https://github.com/autowarefoundation/autoware.universe/pull/8468

They've updated:
- https://github.com/autowarefoundation/sample_sensor_kit_launch/pull/103

But not https://github.com/autowarefoundation/awsim_labs_sensor_kit_launch

pandyah, 2024-10-02:
> @pandyah Can you share the full Autoware log file ? You can find it in `$HOME/.ros/log/{DATE-...}/launch.log`

@maxime-clem, Please find attached launch logs of Autoware.
[launch.log](https://github.com/user-attachments/files/17218516/launch.log)
Thanks

xmfcx, 2024-10-02:
@pandyah from your logs:

> ```
> 1727830261.3353379 [component_container-34] [31m[ERROR 1727830261.331307119] [sensing.lidar.right.pointcloud_preprocessor.pointcloud_container]: Component constructor threw an exception: Statically typed parameter 'distance_ratio' must be initialized. (on_load_node() at ./src/component_manager.cpp:282)[0m
> 1727830261.3424008 [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'ring_outlier_filter' of type 'autoware::pointcloud_preprocessor::RingOutlierFilterComponent' in container '/sensing/lidar/right/pointcloud_preprocessor/pointcloud_container': Component constructor threw an exception: Statically typed parameter 'distance_ratio' must be initialized.
> ```

Both has been today solved by:
- https://github.com/autowarefoundation/awsim_labs_sensor_kit_launch/pull/6

Could you do:

```
cd /your/autoware/folder
vcs import src < autoware.repos
vcs import src < tools.repos
vcs pull src
colcon build --symlink-install --event-handlers=console_cohesion+ --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo
```

And try again?

For me, I had the same problems and it was solved.

pandyah, 2024-10-04:
Thanks very much for the details. I tried the same at my end. Unfortunately, AWSIM still freezes and does not get integrated with Autoware. Please find attached the launch.log

[launch.log](https://github.com/user-attachments/files/17253961/launch.log)

maxime-clem, 2024-10-04:
> [launch.log](https://github.com/user-attachments/files/17253961/launch.log)

```
1728020535.5426111 [component_container_mt-58] [E] [TRT] 1: [runtime.cpp::parsePlan::314] Error Code 1: Serialization (Serialization assertion plan->header.magicTag == rt::kPLAN_MAGIC_TAG failed.)
```
This looks like an issue with the version of your TensorRT library.
Please use the `setup-dev-env.sh` script (in the `autoware` repository) to setup your environment. Otherwise please use Docker.

pandyah, 2024-10-04:
Thanks a lot @maxime-clem  for the prompt response. Tried the `setup-dev-env.sh` from Autoware, removed the build, install and log but looks like it is picking up the Tensorrt version 10.5. Trying to sort out this. If it does not work will switch to docker.

pandyah, 2024-10-07:
> Thanks a lot @maxime-clem for the prompt response. Tried the `setup-dev-env.sh` from Autoware, removed the build, install and log & recompiled but looks like it is picking up the Tensorrt version 10.5. Trying to sort out this. If it does not work will switch to docker.

Thanks all for your help. I am new to AWSIM. Carla works fine with Autoware (Release 2023.10) but does not support the latest release. Therefore  wanted to try AWSIM.

Please find attached latest log file. Now at least it is able to localize the vehicle and shows the point cloud in RVIZ. However, in few seconds the AWSIM freezes with point cloud error. When AWSIM freezes, the CTRL-C does not work and only option is to use kill -9 <PID> to stop AWSIM.
[launch.log](https://github.com/user-attachments/files/17272107/launch.log)

Also, while running only AWSIM without sourcing the ROS2 environment using ./awsim_labs_v1.5.1/awsim_labs.x86_64 without Autoware, the Vehicle Controller keys does not work. Not able to move the vehicle. The hotkeys related to VPPVehicleSignalHandler.cs works fine.

Thanks a lot.

maxime-clem, 2024-10-07:
> AWSIM freezes with point cloud error
Are you running out of memory ? Can you slowdown AWSIM (e.g., to run at 0.3x real time) to see if it prevents this issue ?

pandyah, 2024-10-07:
> > AWSIM freezes with point cloud error
> 
> Are you running out of memory ? Can you slowdown AWSIM (e.g., to run at 0.3x real time) to see if it prevents this issue ?

Thanks very much for the response @maxime-clem 
Tried by reducing the rate to even as low as 0.1x. However, same result. AWSIM freezes in few secs.

Error from Autoware logs
```
[ERROR 1728284131.730102552] [sensing.lidar.left.distortion_corrector_node]: The pointcloud layout is not compatible with PointXYZIRCAEDT. Aborting (isInputValid() at /home/hitesh/autoware_latest/autoware/src/universe/autoware.universe/sensing/autoware_pointcloud_preprocessor/src/distortion_corrector/distortion_corrector.cpp:173)

```
Also, as mentioned earlier, AWSIM running without Autoware freezes after few minutes and there is no response to Vehicle Controller keys.
[image]

[image]

Thanks for the help.

xmfcx, 2024-10-07:
@pandyah I am sorry to hear you are having these troubles. Let's debug it step by step.

> ```
> [ERROR 1728284131.730102552]
> [sensing.lidar.left.distortion_corrector_node]:
> The pointcloud layout is not compatible with PointXYZIRCAEDT.
> Aborting (isInputValid() at [...]

I'm aware of this bug, it is also there for me too. But it didn't prevent me from driving autonomously. We should fix it but it is not urgent for now.

> Also, as mentioned earlier, AWSIM running without Autoware freezes after few minutes and there is no response to Vehicle Controller keys.

Just running AWSIM Labs as it is shouldn't cause it to freeze after a while. We should solve this in isolation first.

Let's make sure ROS 2 is not sourced first:

Run the following in a terminal.
```bash
echo $ROS_DISTRO
```
This should return an empty line.

Then please download and open the new awsim_labs executable from that terminal:

https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.5.2 the new version is 1.5.2 (let's keep going with this version.)

In order for keyboard control to work, you need to click on this button:
[image]

And you should be able to drive with either wasd or arrow keys. And no matter how long it runs, it should not freeze.

I did it on my machine and it seems to be running smoothly by itself.

Could you confirm this much first?

xmfcx, 2024-10-07:
And with the DDS side of things, could you confirm you have these configured correctly?

https://autowarefoundation.github.io/autoware-documentation/main/installation/additional-settings-for-developers/network-configuration/dds-settings/

This is important for running high traffic DDS applications (like Autoware) on your local machine with best performance.

pandyah, 2024-10-08:
> [xmfcx](/xmfcx)

Thanks a lot for all the details.
Have ensured the points you have mentioned regarding ROS2 not sourced, downloaded v1.5.2 and DDS settings

AWSIM running independently works with the arrow keys smoothly and does not freeze. This is great.

However, it freezes when running with Autoware (Irrespective if it is started before Autoware or after Autoware)

Thanks a lot.

xmfcx, 2025-07-30:
This issue got stale, I've confirmed Autoware 0.45.1 works with AWSIM Labs 1.6.1. If it doesn't work for you, please open a new issue and tag me, I'll try to help.