autowarefoundation/AWSIM-Labs #173: The pointcloud layout is not compatible with PointXYZIRCAEDT or PointXYZIRC. Aborting

HiHeXi, 2024-10-25:
Errors encountered when using the latest version of AutoTware

The pointcloud layout is not compatible with PointXYZIRCAEDT. Aborting

HiHeXi, 2024-10-25:
[image]

HiHeXi, 2024-10-25:
[image]

vividf, 2024-10-25:
I think a PR needs to be created to fix the order `PointXYZIRCEADT` to `PointXYZIRCAEDT` in [AWSI](https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/AWSIM/Scripts/Sensors/LiDAR/PointCloudFormatLibrary.cs#L65C1-L75C20)

WilliamFlinchbaugh, 2024-11-01:
@xmfcx I tried out implementing those changes (just replacing every instance of PointXYZIRCEADT with PointXYZIRCAEDT), but I'm still getting the same error from autoware. Is there anything else that needs to be changed?

xmfcx, 2024-11-01:
@vividf @HiHeXi could you try the latest release?

I've fixed it in [v1.5.3](https://github.com/autowarefoundation/AWSIM-Labs/releases/tag/v1.5.3) and tested, it worked for me.

- https://github.com/autowarefoundation/AWSIM-Labs/releases

closed 2024-11-04 by xmfcx