autowarefoundation/AWSIM-Labs #176: Errors encountered when using the latest version of AutoTware

HiHeXi, 2024-11-04:
[autoware_ekf_localizer_node-43] [WARN] [1730684733.088041349] [localization.pose_twist_fusion_filter.ekf_localizer]: EKF period may be too slow to finish pose smoothing!! (0.110012 sec)
[mrm_handler_node-27] [INFO] [1730684733.090250635] [system.mrm_handler]: MRM State changed: MRM_OPERATING -> NORMAL
[mrm_handler_node-27] [WARN] [1730684733.092561258] [system.mrm_handler]: EMERGENCY_STOP is canceled.
[component_container_mt-76] [INFO] [1730684733.092352179] [control.vehicle_cmd_gate]: The operation mode is changed, but the TurnIndicatorsCommand is not received yet:
[autoware_ekf_localizer_node-43] [WARN] [1730684733.192945414] [localization.pose_twist_fusion_filter.ekf_localizer]: EKF period may be too slow to finish pose smoothing!! (0.111061 sec)
[autoware_ndt_scan_matcher_node-39] [INFO] [1730684733.274911475] [localization.pose_estimator.ndt_scan_matcher]: pose_buffer_.size() < 2
[autoware_ndt_scan_matcher_node-39] [INFO] [1730684733.451575034] [localization.pose_estimator.ndt_scan_matcher]: pose_buffer_.size() < 2
[mrm_handler_node-27] [INFO] [1730684733.463253369] [system.mrm_handler]: MRM State changed: NORMAL -> MRM_OPERATING
[mrm_handler_node-27] [WARN] [1730684733.465992061] [system.mrm_handler]: MRM behavior is None. Do nothing.
[mrm_handler_node-27] [WARN] [1730684733.466432386] [system.mrm_handler]: EMERGENCY_STOP is operated.
[component_container_mt-76] [INFO] [1730684733.548544870] [control.vehicle_cmd_gate]: The operation mode is changed, but the TurnIndicatorsCommand is not received yet:
[mrm_handler_node-27] [INFO] [1730684733.550121389] [system.mrm_handler]: MRM State changed: MRM_OPERATING -> MRM_SUCCEEDED
[component_container_mt-76] [INFO] [1730684733.637345635] [control.vehicle_cmd_gate]: The operation mode is changed, but the TurnIndicatorsCommand is not received yet:
[mrm_handler_node-27] [INFO] [1730684733.639854425] [system.mrm_handler]: MRM State changed: MRM_SUCCEEDED -> NORMAL
[mrm_handler_node-27] [WARN] [1730684733.642659612] [system.mrm_handler]: EMERGENCY_STOP is canceled.
[autoware_ndt_scan_matcher_node-39] [INFO] [1730684733.642663457] [localization.pose_estimator.ndt_scan_matcher]: pose_buffer_.size() < 2
[mrm_handler_node-27] [INFO] [1730684733.773529695] [system.mrm_handler]: MRM State changed: NORMAL -> MRM_OPERATING
[mrm_handler_node-27] [WARN] [1730684733.773618682] [system.mrm_handler]: MRM behavior is None. Do nothing.
[mrm_handler_node-27] [WARN] [1730684733.776007653] [system.mrm_handler]: EMERGENCY_STOP is operated.
[component_container_mt-76] [INFO] [1730684733.810114161] [control.vehicle_cmd_gate]: The operation mode is changed, but the TurnIndicatorsCommand is not received yet:
[mrm_handler_node-27] [INFO] [1730684733.811639112] [system.mrm_handler]: MRM State changed: MRM_OPERATING -> NORMAL
[mrm_handler_node-27] [WARN] [1730684733.812089326] [system.mrm_handler]: EMERGENCY_STOP is canceled.

xmfcx, 2024-11-07:
There might be warnings but were you able to drive from A to B?

HiHeXi, 2024-11-11:
> 可能会有警告，但您能从 A 开车到 B 吗？

Cannot drive from A to B

xmfcx, 2024-11-11:
## Logs

Could you share the full logs to help us understand the cause?

For this,
1. Remove the logs in `~/.ros/log`
2. Run the Autoware with AWSIM Labs
3. What do you see? Do you see the localization output?
4. Share the launch log file from `~/.ros/log` here.

## Generic system information

Could you install screenfetch and share the basic system information with us?

1. `sudo apt install screenfetch -y`
2. `screenfetch -n`
3. Share the results.

### NVIDIA Driver and CUDA version

1. Run `nvidia-smi` and share the first 3 lines
> ```
> $ nvidia-smi
> Mon Nov 11 11:05:08 2024       
> +---------------------------------------------------------------------------------------+
> | NVIDIA-SMI 545.23.08              Driver Version: 545.23.08    CUDA Version: 12.3     |
> |-----------------------------------------+----------------------+----------------------+
> ```
> 
2. Run `nvcc -V` and share the results.
> ```
> $ nvcc -V
> nvcc: NVIDIA (R) Cuda compiler driver
> Copyright (c) 2005-2023 NVIDIA Corporation
> Built on Wed_Nov_22_10:17:15_PST_2023
> Cuda compilation tools, release 12.3, V12.3.107
> Build cuda_12.3.r12.3/compiler.33567101_0
> ```

HiHeXi, 2024-11-12:
```
hx@hx:~/.ros/log$ screenfetch -n
/usr/bin/screenfetch: 行 1851: -: 语法错误：需要操作数（错误记号是 "-"）
 hx@hx
 OS: Ubuntu 22.04 jammy
 Kernel: x86_64 Linux 6.8.0-48-generic
 Uptime: 7m
 Packages: 3694
 Shell: bash
 Resolution: 1920x1080
 DE: GNOME 41.7
 WM: Mutter
 WM Theme: Adwaita
 GTK Theme: Yaru [GTK2/3]
 Icon Theme: Yaru
 Font: Ubuntu 11
 Disk: 250G / 885G (30%)
 CPU: 11th Gen Intel Core i7-11800H @ 16x 4.6GHz [66.0°C]
 GPU: NVIDIA GeForce RTX 3050 Ti Laptop GPU
 RAM: -
```

```
hx@hx:~/.ros/log$ nvidia-smi
Tue Nov 12 09:23:56 2024       
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 555.42.06              Driver Version: 555.42.06      CUDA Version: 12.5     |
|----
```

```
hx@hx:~/.ros/log$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Wed_Nov_22_10:17:15_PST_2023
Cuda compilation tools, release 12.3, V12.3.107
Build cuda_12.3.r12.3/compiler.33567101_0
```
[launch.log](https://github.com/user-attachments/files/17709830/launch.log)

[vehicle_velocity_converter_node_6104_1731374480717.log](https://github.com/user-attachments/files/17709832/vehicle_velocity_converter_node_6104_1731374480717.log)
[traffic_light_occlusion_predictor_node_6370_1731374480817.log](https://github.com/user-attachments/files/17709833/traffic_light_occlusion_predictor_node_6370_1731374480817.log)
[traffic_light_multi_camera_fusion_node_6395_1731374480942.log](https://github.com/user-attachments/files/17709834/traffic_light_multi_camera_fusion_node_6395_1731374480942.log)
[traffic_light_map_visualizer_node_6458_1731374481200.log](https://github.com/user-attachments/files/17709835/traffic_light_map_visualizer_node_6458_1731374481200.log)
[traffic_light_map_based_detector_node_6338_1731374480895.log](https://github.com/user-attachments/files/17709837/traffic_light_map_based_detector_node_6338_1731374480895.log)
[topic_state_monitor_node_5931_1731374480507.log](https://github.com/user-attachments/files/17709838/topic_state_monitor_node_5931_1731374480507.log)
[topic_state_monitor_node_5929_1731374480519.log](https://github.com/user-attachments/files/17709839/topic_state_monitor_node_5929_1731374480519.log)
[topic_state_monitor_node_5927_1731374480481.log](https://github.com/user-attachments/files/17709840/topic_state_monitor_node_5927_1731374480481.log)
[topic_state_monitor_node_5925_1731374480456.log](https://github.com/user-attachments/files/17709841/topic_state_monitor_node_5925_1731374480456.log)
[topic_state_monitor_node_5923_1731374480479.log](https://github.com/user-attachments/files/17709842/topic_state_monitor_node_5923_1731374480479.log)
[topic_state_monitor_node_5921_1731374480448.log](https://github.com/user-attachments/files/17709843/topic_state_monitor_node_5921_1731374480448.log)
[topic_state_monitor_node_5919_1731374480475.log](https://github.com/user-attachments/files/17709844/topic_state_monitor_node_5919_1731374480475.log)
[topic_state_monitor_node_5917_1731374480449.log](https://github.com/user-attachments/files/17709845/topic_state_monitor_node_5917_1731374480449.log)
[topic_state_monitor_node_5915_1731374480454.log](https://github.com/user-attachments/files/17709846/topic_state_monitor_node_5915_1731374480454.log)
[topic_state_monitor_node_5913_1731374480455.log](https://github.com/user-attachments/files/17709847/topic_state_monitor_node_5913_1731374480455.log)
[topic_state_monitor_node_5911_1731374480451.log](https://github.com/user-attachments/files/17709848/topic_state_monitor_node_5911_1731374480451.log)
[topic_state_monitor_node_5909_1731374480464.log](https://github.com/user-attachments/files/17709849/topic_state_monitor_node_5909_1731374480464.log)
[topic_state_monitor_node_5907_1731374480464.log](https://github.com/user-attachments/files/17709850/topic_state_monitor_node_5907_1731374480464.log)
[topic_state_monitor_node_5905_1731374480451.log](https://github.com/user-attachments/files/17709851/topic_state_monitor_node_5905_1731374480451.log)
[topic_state_monitor_node_5903_1731374480472.log](https://github.com/user-attachments/files/17709852/topic_state_monitor_node_5903_1731374480472.log)
[shape_estimation_node_6249_1731374480785.log](https://github.com/user-attachments/files/17709853/shape_estimation_node_6249_1731374480785.log)
[service_log_checker_node_5899_1731374480454.log](https://github.com/user-attachments/files/17709854/service_log_checker_node_5899_1731374480454.log)
[rviz2_6782_1731374483141.log](https://github.com/user-attachments/files/17709855/rviz2_6782_1731374483141.log)
[routing_adaptor_node_6745_1731374481446.log](https://github.com/user-attachments/files/17709856/routing_adaptor_node_6745_1731374481446.log)
[robot_state_publisher_5893_1731374480458.log](https://github.com/user-attachments/files/17709857/robot_state_publisher_5893_1731374480458.log)
[relay_6335_1731374480899.log](https://github.com/user-attachments/files/17709858/relay_6335_1731374480899.log)
[python3_6702_1731374482625.log](https://github.com/user-attachments/files/17709860/python3_6702_1731374482625.log)
[python3_5961_1731374482800.log](https://github.com/user-attachments/files/17709862/python3_5961_1731374482800.log)
[python3_5939_1731374482690.log](https://github.com/user-attachments/files/17709863/python3_5939_1731374482690.log)
[python3_4732_1731374459547.log](https://github.com/user-attachments/files/17709864/python3_4732_1731374459547.log)
[... 81 lines cut ...]
> 
> ### NVIDIA 驱动程序和 CUDA 版本
> 
[... cut ...]
> > +---------------------------------------------------------------------------------------+
> > | NVIDIA-SMI 545.23.08              Driver Version: 545.23.08    CUDA Version: 12.3     |
> > |-----------------------------------------+----------------------+----------------------+
> > ```
> 
>     2. 跑步  `nvcc -V` 并分享结果。
> 
> 
> > ```
> > $ nvcc -V
> > nvcc: NVIDIA (R) Cuda compiler driver
> > Copyright (c) 2005-2023 NVIDIA Corporation
> > Built on Wed_Nov_22_10:17:15_PST_2023
> > Cuda compilation tools, release 12.3, V12.3.107
> > Build cuda_12.3.r12.3/compiler.33567101_0
> > ```

HiHeXi, 2024-11-13:
I have uploaded the wrong video on YouTube
https://youtu.be/bXK5hAcwhLU

> Autoware 与 CUDA 12.3 兼容。 与 CUDA 12.3 兼容的 NVIDIA 驱动程序版本是 `545.23.08`。您可能想安装它。 （ `sudo apt install nvidia-driver-545`）（同时清除当前驱动程序） 请参阅此处的相关表格：https: [ //docs.nvidia.com/deploy/cuda-compatibility/#use-the-right-compat-package ](https://docs.nvidia.com/deploy/cuda-compatibility/#use-the-right-compat-package)
> 
> 顺便说一句，我不认为这个问题与 AWSIM 实验室有关。
> 
> 您可以运行以下命令并报告它们是否对您有用吗？
> 
>     1. [https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/ ](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/)
> 
> 
>     * 它仅运行规划、非常基本的感知和系统诊断部分。
> 
>     * 即使没有 GPU，这也能工作。
> 
> 
>     2. [https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/ ](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/)
> 
> 
>     * 这会在 rosbag 文件上运行感知和本地化。
> 
>     * 我还不知道你的系统出了什么问题，但我认为这对你来说也会失败，但让我们尝试一下。
> 
> 
> 如果它们有效，那么我们可以进一步调试。如果没有，我们应该关注他们。

HiHeXi, 2024-11-14:
@xmfcx Thank you for your help. The issue has been resolved

closed 2024-11-14 by HiHeXi