autowarefoundation/AWSIM-Labs #177: Localization is not happening in latest version of AWSim Labs

mustaeen-7, 2024-11-09:
I tried both AWSIM and AWSIM Labs in both localization is not happening in Rviz2 Autoware. Its flickering between initializing to Uninitialized. Can anyone please help me regarding this. I am using ubuntu 22.04 ROS2 Humble. And I cannot see any lidar scans on AWSIM interface.Below I pasted my pc screenshots
[image]
[image]

maxime-clem, 2024-11-10:
Please share the Autoware log file. You can find it in `$HOME/.ros/log/{DATE-USER}/launch.log`.

mustaeen-7, 2024-11-11:
[launch.log](https://github.com/user-attachments/files/17695608/launch.log)
Thanks for the response. Here is the log file.

maxime-clem, 2024-11-11:
> 1731143821.3189173 [component_container_mt-59] [W] [TRT] Unable to determine GPU memory usage
1731143821.3263628 [component_container_mt-59] [W] [TRT] Unable to determine GPU memory usage
1731143821.3278058 [component_container_mt-59] [I] [TRT] [MemUsageChange] Init CUDA: CPU +4, GPU +0, now: CPU 28, GPU 0 (MiB)
1731143821.3286767 [component_container_mt-59] [W] [TRT] CUDA initialization failure with error: 100. Please check your CUDA installation:  http://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html
1731143821.3287909 [component_container_mt-59] [E] [TRT] Fail to create builder
1731143821.8940184 [component_container_mt-1] [W] [TRT] CUDA initialization failure with error: 100

You have an issue with CUDA. Please make sure you have a Nvidia GPU and installed the drivers using the `setup-dev-env.sh` script.

mustaeen-7, 2024-11-15:
I have installed all required drivers. Just to test I launched planning_simulator, in that I could not run the car in Auto mode after setting 2d Pose Estimate and goal point. The path is generating but all other modes are getting disabled. Here are teh screenshots below
[image]
[image]
[image]

mustaeen-7, 2024-11-15:
launch.log file of above error
[launch.log](https://github.com/user-attachments/files/17770630/launch.log)

mustaeen-7, 2024-11-15:
Yes it is happening everytime I run it. I am using latest version of Autoware. I followed this Source installation link:https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/

mustaeen-7, 2024-11-18:
Found out the problem. It was due Nvidia Drivers for Localisation. After removing everything and reinstalling, the issue got fixed. Thanks for the support.

closed 2024-11-18 by mustaeen-7