autowarefoundation/AWSIM-Labs #186: UnsatisfiedLinkError: tier4_simulation_msgs__rosidl_typesupport_c.dll

samanipour, 2025-01-24:
Hi, @mozhoku @SujayAmberkar @Aysenayilmaz @xmfcx if you could help me to solve this issue I would be very appreciated

I'm trying to [setup AWSIM-Lab as Unity Project](https://autowarefoundation.github.io/AWSIM-Labs/main/GettingStarted/SetupUnityProject/) on Windows
### System setup:

- AWSIM-Lab latest [commit on main branch](https://github.com/autowarefoundation/AWSIM-Labs/commit/ceb1c317a5970e7dc19501f202e18c1e8f9e61ab) (also release v1.5.3)
- Windows 11
- [Unity Editor version 2022.3.36f1](https://unity.com/releases/editor/whats-new/2022.3.36#installs)
- Microsoft Visual Studio Community 2022 (Associated build tools with unity editor)

When I run the demo I got the following error, it seems the error raised from 

`UnsatisfiedLinkError: tier4_simulation_msgs__rosidl_typesupport_c.dll
ROS2.DllLoadUtilsWindowsDesktop.ROS2.DllLoadUtils.LoadLibraryNoSuffix (System.String fileName) (at <c2fd60e12ae74b49b3633ef9563fbe65>:0)
tier4_simulation_msgs.msg.DummyObject..cctor () (at <875a1a567ccc4ffe8e2115395ef84ff7>:0)
Rethrow as TypeInitializationException: The type initializer for 'tier4_simulation_msgs.msg.DummyObject' threw an exception.
(wrapper write-barrier) System.Object.wbarrier_conc(intptr)
System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) (at <51fded79cd284d4d911c5949aff4cb21>:0)
Rethrow as TargetInvocationException: Exception has been thrown by the target of an invocation.
System.Reflection.RuntimeConstructorInfo.InternalInvoke (System.Object obj, System.Object[] parameters, System.Boolean wrapExceptions) (at <51fded79cd284d4d911c5949aff4cb21>:0)
System.RuntimeType.CreateInstanceMono (System.Boolean nonPublic, System.Boolean wrapExceptions) (at <51fded79cd284d4d911c5949aff4cb21>:0)
System.RuntimeType.CreateInstanceSlow (System.Boolean publicOnly, System.Boolean wrapExceptions, System.Boolean skipCheckThis, System.Boolean fillCache) (at <51fded79cd284d4d911c5949aff4cb21>:0)
System.RuntimeType.CreateInstanceDefaultCtor (System.Boolean publicOnly, System.Boolean skipCheckThis, System.Boolean fillCache, System.Boolean wrapExceptions, System.Threading.StackCrawlMark& stackMark) (at <51fded79cd284d4d911c5949aff4cb21>:0)
System.Activator.CreateInstance[T] () (at <51fded79cd284d4d911c5949aff4cb21>:0)
ROS2.Subscription`1[T]..ctor (System.String subTopic, ROS2.Node node, System.Action`1[T] cb, ROS2.QualityOfServiceProfile qos) (at <2034886caef046439cfa6ebcd9ed8cd7>:0)
ROS2.Node.CreateSubscription[T] (System.String topic, System.Action`1[T] callback, ROS2.QualityOfServiceProfile qos) (at <2034886caef046439cfa6ebcd9ed8cd7>:0)
ROS2.ROS2Node.CreateSubscription[T] (System.String topicName, System.Action`1[T] callback, ROS2.QualityOfServiceProfile qos) (at Assets/Ros2ForUnity/Scripts/ROS2Node.cs:92)
AWSIM.SimulatorROS2Node.CreateSubscription[T] (System.String topicName, System.Action`1[T] callback, ROS2.QualityOfServiceProfile qos) (at Assets/AWSIM/Scripts/ROS/SimulatorROS2Node.cs:86)
AWSIM.RVIZNPCSpawner.Start () (at Assets/AWSIM/Scripts/NPCs/RvizNPCSpawner/RVIZNPCSpawner.cs:53)`

[image]

I should also mentioned that there are some other errors related to missing RobotecGPULidar which could be solved by the following proposed solution in [this issue](https://github.com/RobotecAI/RobotecGPULidar/issues/330#issuecomment-2598193526).

SujayAmberkar, 2025-01-24:
Hi @samanipour, the error you are facing is in the RVIZNPCSpawnner.cs file over here:
```cs
void Start()
        {
            // Initialize the ROS2 node and create the subscription
            dummyPerceptionSubscriber
                = SimulatorROS2Node.CreateSubscription<tier4_simulation_msgs.msg.DummyObject>(
                        dummyPerceptionTopic, OnObjectInfoReceived, qosSettings.GetQoSProfile());
        }
```
I think this error occurs because it is unable to subscribe to `tier4_simulation_msgs.msg.DummyObject`, likely due to the missing `tier4_simulation_msgs__rosidl_typesupport_c.dll` or its dependencies, as it says. Could you try rebuilding Autoware, since the issue is probably not on the Unity side but rather due to an incomplete Autoware build?

samanipour, 2025-01-24:
@SujayAmberkar Thanks for your response, 
yes the error caused by `RVIZNPCSpawnner.cs` but I think the `tier4_simulation_msgs__rosidl_typesupport_c.dll` should be part of [Ros2ForUnity](https://github.com/autowarefoundation/AWSIM-Labs/tree/main/Assets/Ros2ForUnity), because if you check [`AWSIM-Labs\Assets\Ros2ForUnity\Plugins\Linux\x86_64`](https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/Ros2ForUnity/Plugins/Linux/x86_64/libtier4_simulation_msgs__rosidl_typesupport_c.so) you could see the equivalent library as `libtier4_simulation_msgs__rosidl_typesupport_c.so` for Linux. So I though maybe the `tier4_simulation_msgs__rosidl_typesupport_c.dll` should be exists in [`AWSIM-Labs\Assets\Ros2ForUnity\Plugins\Windows\x86_64`](https://github.com/autowarefoundation/AWSIM-Labs/tree/main/Assets/Ros2ForUnity/Plugins/Windows/x86_64), am I wrong?

SujayAmberkar, 2025-01-24:
@samanipour Ahh yes, you're right. `libtier4_simulation_msgs__rosidl_typesupport_c` must be present in `\windows\x86_64`. No need to rebuild autoware, try reinstalling ros2-for-unity for windows https://github.com/RobotecAI/ros2-for-unity/blob/develop/README-WINDOWS.md I was using linux so it worked for me out of the box.

samanipour, 2025-01-26:
@SujayAmberkar 
Thanks!
I'm curious why `libtier4_simulation_msgs__rosidl_typesupport_c` is not included in the official ros2-for-unity window build release (https://github.com/RobotecAI/ros2-for-unity/releases). 

Given its absence, I'm concerned that following the build instructions you provided (https://github.com/RobotecAI/ros2-for-unity/blob/develop/README-WINDOWS.md) might not generate the DLL we need.

Do you have any insights into why this library is missing or how we can ensure the build process includes it?

SujayAmberkar, 2025-01-27:
@samanipour I'm not really sure about it. Maybe you can ask this question in ros2-for-unity's repo.