autowarefoundation/AWSIM-Labs #190: Using AWSIM Labs With Terrain

RezaHosseini80, 2025-02-26:
Hi everyone, has anyone tested AWSIM Labs with terrain?
I have a large terrain created from a heightmap and texture layers, but I can't use it inside AWSIM Labs. I believe the issue is related to the RGL project, which simulates LiDAR. It seems to disable the LiDAR sensor in the scene and displays the following errors:

"
RGLException: cuda error: out of memory (code=2) @ /home/mateusz/RobotecGPULidar/src/memory/MemoryOperations.hpp:65
RGLUnityPlugin.RGLNativeAPI.CheckErr (System.Int32 status) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:244)
RGLUnityPlugin.RGLNativeAPI.GraphGetResult[T] (System.IntPtr node, RGLUnityPlugin.RGLField field, T[]& data, System.Int32 expectedPointSize) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:524)
RGLUnityPlugin.RGLNodeSequence.GetResultData[T] (T[]& data) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNodeSequence.cs:436)
RGLUnityPlugin.PointCloudVisualization.OnNewLidarData () (at Assets/RGLUnityPlugin/Scripts/PointCloudVisualization.cs:199)
RGLUnityPlugin.LidarSensor.NotifyNewData () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:296)
RGLUnityPlugin.LidarSensor.FixedUpdate () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:257)
"

"
RGLException: A previous unrecoverable error has corrupted RobotecGPULidar's internal state. Disabling LidarSensor components. The application must be restarted!
RGLUnityPlugin.RGLNativeAPI.CheckErr (System.Int32 status) (at Assets/RGLUnityPlugin/Scripts/LowLevelWrappers/RGLNativeAPI.cs:244)
RGLUnityPlugin.SceneManager.SynchronizeSceneTime () (at Assets/RGLUnityPlugin/Scripts/SceneManager.cs:199)
RGLUnityPlugin.SceneManager.DoUpdate (System.Int32 fixedUpdateFrame) (at Assets/RGLUnityPlugin/Scripts/SceneManager.cs:138)
RGLUnityPlugin.LidarSensor.Capture () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:324)
RGLUnityPlugin.LidarSensor.FixedUpdateLogic () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:290)
RGLUnityPlugin.LidarSensor.FixedUpdate () (at Assets/RGLUnityPlugin/Scripts/LidarSensor.cs:249)
"

and after this needs to restart unity to use RGL again anywhere in project 
i tried this with Velodyne VLS128 and another lidar i created from manual and the result is same.

Has anyone encountered this issue or found a solution?
and if we can't use terrain in this project, how can i convert the terrain to a model witch AWSIM Labs Supporting?

zubxxr, 2025-03-01:
Hi, concerning the ```RGLException: A previous unrecoverable error has corrupted RobotecGPULidar's internal state. Disabling LidarSensor components. The application must be restarted!```, I have had this issue multiple times, and I have found a fix for it. However, I have yet to find an easier fix. Unfortunately, the fix requires you to reinstall your NVIDIA driver and reboot your computer, and the only issue with it is that it is time consuming.

These are the steps I followed. I personally have ```nvidia-driver-565```, but you should run ```nvidia-smi``` initially to get your version and reinstall that version. So for me I would run ```sudo apt install -y nvidia-driver-565```.

```bash
  nvidia-smi
  sudo apt purge -y nvidia-*
  sudo add-apt-repository ppa:graphics-drivers/ppa 
  sudo apt update 
  sudo apt install -y nvidia-driver-<your-version>
  nvidia-smi
  sudo reboot
```

Let me know if this works or if you can find an even easier solution!

WilliamFlinchbaugh, 2025-05-27:
I just had the same issue getting the `RGLException: A previous unrecoverable error has corrupted RobotecGPULidar's internal state. Disabling LidarSensor components. The application must be restarted!`. 

The issue arose for me after leaving Unity open and running for a while (forgot to turn it off), and then I had to eventually kill the process, and @zubxxr's solution of reinstalling nvidia drivers worked for me.

I want to add that, assuming you're running autoware, you may have problems with the held back packages and stuff breaking with cuda and tensorrt.

The full fix for me looked like this:

```
sudo apt-mark unhold  \
  "cuda*"             \
  "libcudnn*"         \
  "libnvinfer*"       \
  "libnvonnxparsers*" \
  "libnvparsers*"     \
  "tensorrt*"         \
  "nvidia*"

sudo apt purge -y       \
  "cuda*"             \
  "libcudnn*"         \
  "libnvinfer*"       \
  "libnvonnxparsers*" \
  "libnvparsers*"     \
  "tensorrt*"         \
  "nvidia*"

sudo apt autoremove
sudo add-apt-repository ppa:graphics-drivers/ppa
sudo apt update
sudo apt install -y nvidia-driver-<your-version>
sudo reboot
```

Then, after rebooting, follow the manual installation for cuda [here](https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/cuda) and tensorrt [here](https://github.com/autowarefoundation/autoware/tree/main/ansible/roles/tensorrt)

I think that this issue can likely be closed.