autowarefoundation/AWSIM-Labs #193: ROS2 use-sim_time

mohamedleithy, 2025-05-04:
Hello AWSIM-labs Team,

I am currently using AWSIM with ROS 2 and have encountered a challenge with simulation time management. Specifically, I am interested in enabling the use of the ROS 2 parameter "use_sim_time" so that the simulator properly subscribes to an externally published clock on the /clock topic.

Appreciate your support on this.

maxime-clem, 2025-05-05:
> so that the simulator properly subscribes to an externally published clock

If I understand correctly, you want to set `TimeSource: "ROS2"` (see https://tier4.github.io/AWSIM/Components/Clock/ClockPublisher/).

mohamedleithy, 2025-05-05:
The ROS2 clock system primarily relies on the system time by default since the use_sim_time parameter is not configured. Normally this parameter can be set either through launch files or via ROS2 launch commands (e.g., ros2 run your_package your_node --ros-args -p use_sim_time:=true). In the source code particularly in [ROS2TimeSource.cs](https://github.com/autowarefoundation/AWSIM-Labs/blob/main/Assets/Ros2ForUnity/Scripts/Time/ROS2TimeSource.cs), there's this comment 

`// Create clock which uses system time by default (unless use_sim_time is set in ros2)`. However, I'm not sure how to set this parameter through the Ros2ForUnity package.

maxime-clem, 2025-05-05:
You are right and I also cannot find how to set `use_sim_time`. Maybe the feature is missing from `Ros2ForUnity` 🤔