autowarefoundation/AWSIM-Labs #206: e2e sim on autoware won't launch

Hensbag, 2025-07-17:
When I trying to launch e2e sim I encounter this problem how can I resolve it ? I am following the instructions from awsim labs 
`ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=awsim_labs_vehicle sensor_model:=awsim_labs_sensor_kit map_path:=/home/hensbag/autoware_map/nishishinjuku_autoware_map launch_vehicle_interface:=true`

```
[[INFO] [launch]: All log files can be found below /home/hensbag/.ros/log/2025-07-17-03-30-46-412311-hensbag-ZERNN-5296
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Included launch description missing required argument 'pointcloud_container_name' (description: 'name of pointcloud container'), given: []
Future exception was never retrieved
future: <Future finished exception=InvalidHandle('cannot use Destroyable because destruction was requested')>
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 197, in _load_in_sequence
    self._load_node(next_load_node_request, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
    while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 176, in wait_for_service
    while self.context.ok() and not self.service_is_ready() and timeout_sec > 0.0:
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 159, in service_is_ready
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
Future exception was never retrieved
future: <Future finished exception=InvalidHandle('cannot use Destroyable because destruction was requested')>
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 197, in _load_in_sequence
    self._load_node(next_load_node_request, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
    while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 176, in wait_for_service
    while self.context.ok() and not self.service_is_ready() and timeout_sec > 0.0:
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 159, in service_is_ready
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
Future exception was never retrieved
future: <Future finished exception=InvalidHandle('cannot use Destroyable because destruction was requested')>
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 197, in _load_in_sequence
    self._load_node(next_load_node_request, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
    while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 176, in wait_for_service
    while self.context.ok() and not self.service_is_ready() and timeout_sec > 0.0:
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 159, in service_is_ready
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
Future exception was never retrieved
future: <Future finished exception=InvalidHandle('cannot use Destroyable because destruction was requested')>
Traceback (most recent call last):
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 197, in _load_in_sequence
    self._load_node(next_load_node_request, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/load_composable_nodes.py", line 119, in _load_node
    while not self.__rclpy_load_node_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 176, in wait_for_service
    while self.context.ok() and not self.service_is_ready() and timeout_sec > 0.0:
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 159, in service_is_ready
    with self.handle:
rclpy._rclpy_pybind11.InvalidHandle: cannot use Destroyable because destruction was requested
[INFO] [component_container_mt-1]: process started with pid [5703]
[INFO] [robot_state_publisher-2]: process started with pid [5705]
[INFO] [autoware_raw_vehicle_cmd_converter_node-3]: process started with pid [5707]
[INFO] [autoware_duplicated_node_checker_node-4]: process started with pid [5709]
[INFO] [processing_time_checker_node-5]: process started with pid [5711]
[INFO] [service_log_checker_node-6]: process started with pid [5713]
[INFO] [component_container-7]: process started with pid [5715]
[... 145 lines cut ...]
[autoware_mrm_handler_node-28] [INFO] [1752712263.983818850] [rclcpp]: signal_handler(signum=2)
[component_container_mt-29] [INFO] [1752712263.996401583] [rclcpp]: signal_handler(signum=2)
[autoware_map_projection_loader_node-31] [INFO] [1752712264.013793645] [rclcpp]: signal_handler(signum=2)
[INFO] [component_container-24]: process has finished cleanly [pid 5757]
[INFO] [autoware_topic_state_monitor_node-16]: process has finished cleanly [pid 5733]
[INFO] [autoware_topic_state_monitor_node-12]: process has finished cleanly [pid 5725]
[INFO] [autoware_topic_state_monitor_node-19]: process has finished cleanly [pid 5739]
[INFO] [autoware_topic_state_monitor_node-21]: process has finished cleanly [pid 5747]
[INFO] [autoware_topic_state_monitor_node-22]: process has finished cleanly [pid 5749]
[INFO] [component_container-23]: process has finished cleanly [pid 5754]
[INFO] [autoware_topic_state_monitor_node-20]: process has finished cleanly [pid 5744]
[autoware_map_projection_loader_node-31] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[autoware_map_projection_loader_node-31]   what():  could not create publisher: rcl node's context is invalid, at ./src/rcl/node.c:428
[WARNING] [launch_ros.actions.load_composable_nodes]: Abandoning wait for the '/system/component_state_monitor/container/_container/load_node' service response, due to shutdown.
[ERROR] [autoware_map_projection_loader_node-31]: process has died [pid 5805, exit code -6, cmd '/home/hensbag/autoware/install/autoware_map_projection_loader/lib/autoware_map_projection_loader/autoware_map_projection_loader_node --ros-args -r __ns:=/map -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7 --params-file /tmp/launch_params_sgcx2dhw'].
[INFO] [autoware_diagnostic_graph_aggregator_node-25]: process has finished cleanly [pid 5760]
[INFO] [autoware_hazard_status_converter_node-27]: process has finished cleanly [pid 5773]
[INFO] [component_container_mt-29]: process has finished cleanly [pid 5787]
[INFO] [autoware_mrm_handler_node-28]: process has finished cleanly [pid 5784]
[INFO] [map_hash_generator-30]: process has finished cleanly [pid 5797]
[INFO] [component_state_diagnostics-26]: process has finished cleanly [pid 5764]
```
Here is the log file 

[launch.log](https://github.com/user-attachments/files/21269488/launch.log)

maxime-clem, 2025-07-17:
I could reproduce the issue but it is fixed when using the nightly branches (`vcs import src < autoware-nightly.repos`).

Hensbag, 2025-07-17:
> I could reproduce the issue but it is fixed when using the nightly branches (`vcs import src < autoware-nightly.repos`).

will try thank you