autowarefoundation/AWSIM-Labs #207: Self-driving Fails or Frequently Enters EMERGENCY_STOP

shahriar0651, 2025-07-18:
### Self-driving Fails or Enters EMERGENCY_STOP Due to Pose Discrepancy Between Autoware and AWSIM

#### 🧩 Summary

In my setup with Autoware and AWSIM, I'm encountering inconsistent and undesired behavior when attempting to engage self-driving. The issue appears to stem from a discrepancy between the ego vehicle's pose in AWSIM and the pose used by Autoware for planning and localization.

---

#### ▶️ Case 1: Starting from Origial Spawned Position + Setting Only 2D Goal Pose

If I **do not change the default `2D Pose Estimate` and only specify a `2D Goal Pos`
**:

* **Autoware successfully plans a path** and self-driving works.
* However, the system enters **intermittent `EMERGENCY_STOP` states**.
* Logs show:

  * `mrm_handler` repeatedly toggles `EMERGENCY_STOP`
  * `/vehicle_cmd_gate` reports `Emergency!`
  * Localization nodes report instability (e.g., insufficient pose buffers, delayed twist data).

This suggests that localization is technically running but is unstable, likely due to partial misalignment or sensor sync issues.

---

#### 🚫 Case 2: Setting a 2D Initial Pose Estimate

If **I change** `2D Pose Estimate` estimate in RViz**:

* The ego vehicle **updates its position correctly in AWSIM**.
* However, **Autoware still uses the original spawn location** as the starting point for planning.
* When a goal is sent:

  * The path is incorrectly generated from the initial (pre-pose-estimate) location.
  * **The vehicle never moves** because the pose in AWSIM and the pose used by Autoware are desynchronized.
  * Autodrive does not engage and reports issues such as:

    * Steering not converged
    * Emergency state active
    * No proper localization convergence (e.g., NDT matcher fails with `pose_buffer_.size() < 2`)

This mismatch between AWSIM's updated ego position and Autoware's internal localization state seems to be the root cause of the failure.

---
Versions I used:

AWSIM Demo: v1.6.0
Autoware: Main branch (July 18, 2025/Commit a467517)

maxime-clem, 2025-07-22:
> Case 1

I do not experience this issue so I wonder if it is a performance problem. Do you experience the same issues when reducing the time scale ?

> Case 2

After changing the pose, you need to re-initialize the localization using the `Initialize with GNSS` button in Rviz.

shahriar0651, 2025-07-25:
Thanks, @maxime-clem. Both problems are solved with your recommended solutions. I also believe the issues were there due to hardware limitations (performance problem).