autowarefoundation/AWSIM-Labs #210: Can't open Unity Project

leonidaas, 2025-07-29:
I've tried to open the Unity Project and get compilation errors.
**OS**: Ubuntu 22.04 and Windows 11 Home

**Problem:**
Can't open the project due to compilation errors.

Unity Version: 2022.3.62f1

p2256, 2025-07-30:
Hi there, 

Are you seeing the same errors as this screenshot?

[image]

If so, we are experiencing the same issue. I am also running the same unity version, with lbssl installed on PopOS 22.04, such as the documentation suggests (link: https://autowarefoundation.github.io/AWSIM-Labs/main/GettingStarted/SetupUnityProject/ )

If anyone knows a way to debug this, please reach out!

Thanks,
Cole

p2256, 2025-07-30:
This cleared my errors. Hopefully @leonidaas can confirm it helps their issues as well and can close this thread. 

Thanks @xmfcx for your help, it is greatly appreciated.